Ultimaker/Cura #5880: [4.1.0] Objects cannot be selected after slicing and then switching back to 'Prepare'

rreilink, 2019-06-09:
**Application Version**
4.1.0

**Platform**
macOS

**Printer**
Ultimaker S5 / Ultimaker 3; occurs with either

**Actual Results**
Add a part to the build plate. Slice. Switch to 'Preview'. Switch back to 'Prepare'. Click an object. The object cannot be selected. The object can be selected by 'Select All Models' function.

**Expected results**
Clicking the model should select the model

Ghostkeeper, 2019-06-10:
It's working fine here. Here is me performing your reproduce steps: http://dulek.net/work/SelectingAfterSlicing.mp4

Maybe MacOS only?

rreilink, 2019-06-18:
Indeed I am not experiencing this on Windows. There seems to be some region in the viewport that is'not-clickable':
- Some regions allow selection of the model, others are 'not-clickable'
- When rotating the view, the same region (referred to screen coordinates) stays 'not-clickable'; i.e. changing the view does allow the model to be selected by moving it out of the viewport.

Is there a way to display or save the selection buffer?

fieldOfView, 2019-06-18:
I have seen something similar on Windows, but could not reliably reproduce it and am often doing "naughty" things with the viewport so I was blaming myself. On Windows you would use renderdoc.org to inspect opengl texturebuffers such as the selectionbuffer.

Ghostkeeper, 2019-06-18:
> There seems to be some region in the viewport that is'not-clickable'

Yeah we've seen this quite often but never could lay a finger on what's causing it. And the bug is very fleeting. Sometimes it gets fixed by rotating the camera or something.

Ghostkeeper, 2020-06-25:
We haven't been able to reproduce this issue yet. Needs testing on MacOS though, which is difficult to come by since none of our developers actively use MacOS.

Ghostkeeper, 2020-06-25:
:angry:

Vandrasc, 2020-06-26:
I cannot reproduce either on os x, the models can be successfully selected when clicking on the models. I propose to close this ticket.

Ghostkeeper, 2020-06-30:
@rreilink told me last week via Teams that he's still experiencing this bug.

rreilink, 2020-07-06:
The bug only occurs occasionally. I have updated the description to incorporate this.

Ghostkeeper, 2020-07-07:
Those are the hardest type of bugs to find!

closed 2024-12-07 by github-actions[bot]