Ultimaker/Cura #5881: [4.0-4.1] G10: Retract Issue with RepRap machine selection

PDBeal, 2019-06-10:
**Application Version**
Cura 3.6
Cura 4.0
Cura 4.1

**Platform**
Mac OS - Mojave

**Printer**
CoreXY RepRap printer running on Duet 0.8.5 board running RepRapFirmware-1.23 (DC42 flavor) Firmware in Custom FFF printer set to RepRap. Using Firmware Retraction options in Cura due to machine being E3D Cyclops 2-In 1-Out.

**Actual Results**
Slicing any file with the option Firmware Retraction selected uses "G10 S0" instead of "G10".
"G10 S0" on RepRapFirmware sets the current selected tool's temperature to 0.

**Expected results**
According to The [RepRap G-Code Wiki](https://reprap.org/wiki/G-code#G10:_Retract) a firmware retraction should just be "G10" unless the control board is using firmware repetier at which it would need the "G10 S0" or "G10 S1".

**Additional Information**
This was semi-addressed/acknowledged at the [Duet3d Forums](https://forum.duet3d.com/topic/8505/bug-in-duet-0-8-5-firmware-1-23-with-g10-retract) but the end result never seemed to make it into Cura 4.0 or Cura 4.1.

Ghostkeeper, 2019-06-10:
> "G10 S0" on RepRapFirmware sets the current selected tool's temperature to 0.

[image]

[Yeah](https://reprap.org/wiki/G-code#G10:_Set_tool_Offset_and.2For_workplace_coordinates_and.2For_tool_temperatures). Looks like we need an exception in CuraEngine's g-code export for firmware retractions when using the RepRap g-code flavour.

smartavionics, 2019-06-10:
In the current master branch it does not output G10 S? unless the gcode flavour is repetier.

I am using firmware retraction with Duet controllers and it's working just fine, only G10 and G11 get output.

smartavionics, 2019-06-10:
FYI https://github.com/Ultimaker/CuraEngine/commit/00bc58391afd6b94b4112d3492d5bd8adba013a5

PDBeal, 2019-06-11:
Checking the 4.1 version, it appears as it has been fixed.  I know it has been an issue in 4.0 and 3.6, but  it appears as though it does work properly in 4.1.

closed 2019-06-11 by PDBeal