Ultimaker/Cura #5882: Cura 4.1.0 on Gentoo - segmentation fault after Loadscreen

patchMonkey156, 2019-06-10:
**Application Version**
Cura-4.1.0 (Cura-3.6.0 tested from repositories as well, identical issues found)
Perhaps each versions release notes should note specifically supported versions of Qt and PyQt? Because recklessly changing which version of PyQt causes a cascade of rebuilding/headaches for anyone also running Firefox.

**Platform**
Gentoo Linux 64-bit.
kernel-5.1.8
PyQt - 5.10-r1
Qt - 5.12.3
added an additional local overlay to the portage tree to host updated Cura-specific ebuilds. Using v3.6 from the portage tree proved fruitless, having nearly identical results.

**Printer**
no printer installed yet. Have one on order - will need heavy configuration. Hoping to get this configured and installed first, to streamline the process.

**Actual Results**
the small loading screen runs. then, segmentation fault when that finishes.

**Expected results**
running gui

**Debug Output**
cura --debug
/usr/lib64/python3.6/site-packages/UM/PluginRegistry.py:4: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
[MainThread] UM.Resources.__initializeStoragePaths [416]: Initializing storage paths
[MainThread] UM.Resources.__initializeStoragePaths [427]: Config storage path is /root/.config/cura/4.1
[MainThread] UM.Resources.__initializeStoragePaths [435]: Data storage path is /root/.local/share/cura/4.1
[MainThread] UM.Resources.__initializeStoragePaths [447]: Cache storage path is /root/.cache/cura/4.1
[MainThread] UM.PackageManager.__init__ [52]: Found bundled packages JSON file: /usr/share/uranium/resources/bundled_packages/uranium.json
[MainThread] UM.PackageManager.__init__ [52]: Found bundled packages JSON file: /usr/share/cura/resources/bundled_packages/cura.json
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [105]: Trying OpenGL context 4.1...
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [115]: Yay, we got at least OpenGL 4.1 core: 4.5 Core profile
[MainThread] UM.View.GL.OpenGLContext.detectBestOpenGLVersion [162]: OpenGL renderer type for this OpenGL version: Radeon RX 580 Series (POLARIS10, DRM 3.30.0, 5.1.7-hardopt, LLVM 8.0.0)
[MainThread] UM.Qt.QtApplication.initialize [141]: Detected most suitable OpenGL context version: 4.1 Core profile
[MainThread] UM.Qt.QtApplication.initialize [148]: Initializing job queue ...
[MainThread] UM.Qt.QtApplication.initialize [152]: Initializing version upgrade manager ...
[MainThread] UM.Settings.ContainerRegistry.addContainer [387]: Container [empty_definition_changes] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [387]: Container [empty_variant] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [387]: Container [empty_material] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [387]: Container [empty_quality] added.
[MainThread] UM.Settings.ContainerRegistry.addContainer [387]: Container [empty_quality_changes] added.
[MainThread] UM.PackageManager._loadManagementData [122]: Loaded bundled packages data from /usr/share/uranium/resources/bundled_packages/uranium.json
[MainThread] UM.PackageManager._loadManagementData [122]: Loaded bundled packages data from /usr/share/cura/resources/bundled_packages/cura.json
[MainThread] UM.PackageManager._loadManagementData [139]: Loaded user packages management file from /root/.local/share/cura/4.1/packages.json
[MainThread] UM.PackageManager._saveManagementData [207]: Package management file /root/.local/share/cura/4.1/packages.json was saved
[MainThread] UM.PluginRegistry.initializeBeforePluginsAreLoaded [90]: Loading plugin configuration file '/root/.config/cura/4.1/plugins.json'
2019-06-10 14:05:33,847 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ConsoleLogger
2019-06-10 14:05:33,875 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.__init__ [78]: Found CuraEngine at: /usr/bin/CuraEngine
2019-06-10 14:05:33,877 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin CuraEngineBackend
2019-06-10 14:05:33,885 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin OBJWriter
2019-06-10 14:05:33,889 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin STLWriter
2019-06-10 14:05:33,893 - WARNING - [MainThread] STLReader.STLReader.<module> [26]: Could not find numpy-stl, falling back to slower code.
2019-06-10 14:05:33,896 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin STLReader
2019-06-10 14:05:33,899 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin OBJReader
2019-06-10 14:05:33,903 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin LocalContainerProvider
2019-06-10 14:05:33,910 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin UpdateChecker
2019-06-10 14:05:33,910 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin ConsoleLogger was already loaded
2019-06-10 14:05:33,911 - INFO - [Thread-1] UpdateChecker.UpdateCheckerJob.run [36]: Checking for new version of cura
2019-06-10 14:05:33,919 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin LocalFileOutputDevice
2019-06-10 14:05:33,924 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SimpleView
2019-06-10 14:05:33,931 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin RotateTool
2019-06-10 14:05:33,936 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin MirrorTool
2019-06-10 14:05:33,943 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin ScaleTool
2019-06-10 14:05:33,947 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SelectionTool
2019-06-10 14:05:33,953 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin TranslateTool
2019-06-10 14:05:33,958 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin CameraTool
2019-06-10 14:05:33,962 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin FileLogger
2019-06-10 14:05:33,970 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin PostProcessingPlugin
2019-06-10 14:05:33,976 - INFO - [MainThread] UM.PluginRegistry.loadPlugin [405]: Loaded plugin SolidView
2019-06-10 14:05:33,976 - WARNING - [MainThread] UM.PluginRegistry.loadPlugin [353]: Plugin CuraEngineBackend was already loaded
[... 248 lines cut ...]
  File "/usr/lib64/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib64/python3.6/threading.py", line 884 in _bootstrap

Thread 0x000071360e7fc700 (most recent call first):
  File "/usr/lib64/python3.6/threading.py", line 295 in wait
  File "/usr/lib64/python3.6/threading.py", line 426 in acquire
  File "/usr/lib64/python3.6/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib64/python3.6/site-packages/UM/JobQueue.py", line 112 in run
  File "/usr/lib64/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib64/python3.6/threading.py", line 884 in _bootstrap

Thread 0x000071360effd700 (most recent call first):
  File "/usr/lib64/python3.6/threading.py", line 295 in wait
  File "/usr/lib64/python3.6/threading.py", line 426 in acquire
  File "/usr/lib64/python3.6/site-packages/UM/JobQueue.py", line 86 in _nextJob
  File "/usr/lib64/python3.6/site-packages/UM/JobQueue.py", line 112 in run
  File "/usr/lib64/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib64/python3.6/threading.py", line 884 in _bootstrap

Current thread 0x00007136531d5740 (most recent call first):
  File "/usr/lib64/python3.6/site-packages/UM/Qt/QtApplication.py", line 265 in initializeEngine
  File "/usr/lib64/python3.6/site-packages/cura/CuraApplication.py", line 858 in runWithGUI
  File "/usr/lib64/python3.6/site-packages/cura/CuraApplication.py", line 779 in run
  File "/usr/bin/cura", line 136 in <module>
Segmentation fault

Ghostkeeper, 2019-06-11:
> Perhaps each versions release notes should note specifically supported versions of Qt and PyQt?

We do this through tags in our cura-build-environment repository: https://github.com/Ultimaker/cura-build-environment/tags
Build the build environment with the correct tag, so that the correct versions of Qt and all other dependencies get installed. Or use the scripts in the `./projects` folder to find the version numbers if you want to install them manually. In particular we're using Qt 5.10.1 and PyQt 5.10 for the official builds. We can't upgrade to newer versions of Qt because that would drop support for older MacOS versions.

As for the segfault itself, perhaps you should run Cura through GDB. The stack trace says it's in `initializeEngine` which starts up the QML engine, so that doesn't tell us much.

Also you might want to install zeroconf in your Python environment, but that doesn't cause this crash.

patchMonkey156, 2019-06-11:
Either I'm not as familiar with GDB as I should be:
gentoo-pc$ gdb cura
[ . . . nothing relevant . . . ]
"/usr/bin/cura": not in executable format: File format not recognized
Hmmm... it is set up to work for python3.6, this is weird.

-Would initializeEngine freak out if I'm in Qt-5.12.3, even with the PyQt frontend at the right version?
-Does initializeEngine touch anything related to SSL/TLS? if so, where?

2019-06-10 14:05:37,460 - DEBUG - [MainThread] cura.Machines.Models.BuildPlateModel._update [33]: Updating BuildPlateModel.
Fatal Python error: Segmentation fault
Thread 0x00007135d1db6700 (most recent call first):
  File "/usr/lib64/cura/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 76 in _updateThread
  File "/usr/lib64/python3.6/threading.py", line 864 in run
  File "/usr/lib64/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/lib64/python3.6/threading.py", line 884 in _bootstrap

This has me suspicious. But on a different machine, without a printer plugged in, this runs fine. Just... wow the older graphics card on that one has Slicing take *Forever*.

Ghostkeeper, 2019-06-12:
> Would initializeEngine freak out if I'm in Qt-5.12.3, even with the PyQt frontend at the right version?

Well normally not. I'm using Qt 5.12.2 and PyQt 5.12.1 when running from source and not getting this crash. The only problems I'm having with it is that some elements have the wrong height.

> Does initializeEngine touch anything related to SSL/TLS? if so, where?

No, it shouldn't. It loads up the QML rendering engine. We do use Qt in order to make network requests (like when checking for updates) so that will happen at the same time or even before the QML engine is initialised.

> This has me suspicious. But on a different machine, without a printer plugged in, the Ubuntu PPA version of Cura runs fine. Just... wow the older graphics card on that one has Slicing take Forever.

While Cura is running there are a bunch of threads dedicated to watching USB ports. They will be idling in that `_updateThread` function. This is always the case, regardless of whether there is a device plugged in or not. I don't think it has anything to do with this crash. The other part of that log though, it crashing right when updating the BuildPlateModel, is more telling.

patchMonkey156, 2019-06-13:
Sorry for leaving this to sit for a while, work was suddenly demanding.

You mentioned a ./projects folder to look through for some manual install scripts, but I'm not finding it in the source for either Cura or CuraEngine, nor in the project team's github repos. It sounded like a serious possibility.

As for updating the BuildPlateModel - grep-ing the code has produced some results. I may look into those.

I'm more interested in examining the __init__.py inside of cura and a few other details from the segfault dump itself.

-those lines within the plugins listed? "time.sleep(5)" each time - not likely candidates, from the look of it. Though, I am curious about one small detail...

Though, the one about removeable drives looks interesting. Unless I'm wrong, the thread never made it to the LinuxRemovableDrivePlugin.py - it was too busy sleeping.
 
-the next couple about exception handling - is this how we got the debug log in the first place? But interestingly enough, those files are owned by Uranium-4.1.0 - all the way down to QtApplication.py! That last one is the only line in the dump where nothing included either "wait" or "sleep". I'm gonna go out on a limb here...

>self._qml_engine.load(self._main_qml)

which then, in the same file:

> 96             self._main_qml = "main.qml" #type: str

I feel bad for having to run away after such a quick overview, but back to the grindstone! I'm mostly posting this here as notes. Its so rushed, I'm aware that I might even be barking up the wrong tree.

patchMonkey156, 2019-06-14:
That fixed it. Checking other applications now.

closed 2019-06-14 by patchMonkey156

patchMonkey156, 2019-06-14:
Had to rebuild one to catch it up to the new config. Everything works.