Ultimaker/Cura #5883: I want to request "only retract when crossing open space"

Silithas, 2019-06-11:
Simplify 3d has an option where you can choose to only retract when moving from one print area to another when traveling over open space whether it be to another model on the same bed or say to a different part of the same model. 

I want to request such a feature. If it already exists, please name it.

Block137, 2019-06-11:
This
[image]

smartavionics, 2019-06-11:
The closest that cura can provide is setting combing mode to all.

Ghostkeeper, 2019-06-11:
There are probably some detailed differences between combing mode `All` and the way that Simplify3D implemented it, like support being a separate setting or whether or not it needs to retract when crossing draft shields and prime towers and whatnot. But the goal is to provide good print quality, not to be the same as Simplify3D. So I think then the current state of this feature is as Silithas requested it.

closed 2019-06-11 by Ghostkeeper