Ultimaker/Cura #5884: Cannot find definitions (*.def.json ) directory on Linux but .cfg found

wuyuanyi135, 2019-06-11:
**Application Version**

Cura 4.1.0 (Appimage for Linux)
**Platform**
```
$ uname -a
Linux wuyuanyi-workstation 5.0.12-050012-generic #201905040834 SMP Sat May 4 08:37:03 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

**Printer**
Not related

**Actual Results**
Trying to modify the setting for dual extrusion according to #3451. However, I could not find the `.def.json` on my installation. Followed [this wiki](https://github.com/Ultimaker/Cura/wiki/Definition-Files-Explained) but I only have a bunch of `.cfg` files in the specified directory
```
~/.local/share/cura/4.1/extruders$ ls
creality_cr10_extruder_0+%232.extruder.cfg  custom_extruder_5+%232.extruder.cfg
custom_extruder_1+%232.extruder.cfg         custom_extruder_6+%232.extruder.cfg
custom_extruder_2+%232.extruder.cfg         custom_extruder_7+%232.extruder.cfg
custom_extruder_3+%232.extruder.cfg         custom_extruder_8+%232.extruder.cfg
custom_extruder_4+%232.extruder.cfg         fdmextruder+%232.extruder.cfg
```
Are they the alternatives to the .def.json files? How could I apply "machine_extruder_start_pos_abs" and "machine_extruder_end_pos_abs" with these .cfg files?

**Expected results**
Found the .def.json files

**Additional Information**

smartavionics, 2019-06-11:
The .def.json files are contained within the AppImage. While the AppImage is being executed, the contents are temporarily mounted so you can access them. Alternatively, you can unpack the AppImage.
```
ls /tmp/.mount_Cura-mP3Mpdh/usr/bin/resources
bundled_packages  definitions  extruders  firmware  i18n  images  materials  meshes  qml  quality  setting_visibility  shaders  texts  themes  variants
```

Ghostkeeper, 2019-06-11:
In the AppImage, search for `resources/definitions` and `resources/extruders` to find .def.json files.

wuyuanyi135, 2019-06-11:
Thanks @smartavionics and @Ghostkeeper ! Is the change in the temporarily mounted path persistent?

smartavionics, 2019-06-11:
> Is the change in the temporarily mounted path persistent?

Not 100% sure but I don't think so

wuyuanyi135, 2019-06-11:
@smartavionics In that case, do you have any suggestion how I can customize the extruder definition?

smartavionics, 2019-06-11:
I'm not really the person to answer that as I know next to nothing about how all the frontend configuration stuff works but what I have done is to copy the printer and extruder definitions into the .local tree with different names from the originals and then that will appear as a different printer type. Hope this helps.

wuyuanyi135, 2019-06-11:
@smartavionics Sounds good. Will try it!

closed 2019-06-11 by wuyuanyi135

wuyuanyi135, 2019-06-12:
@Ghostkeeper Is this inconsistent config directory behavior desired for different platforms? I could not find the documentation for .cfg files.

Ghostkeeper, 2019-06-12:
> what I have done is to copy the printer and extruder definitions into the .local tree with different names from the originals and then that will appear as a different printer type. Hope this helps.

That will work as long as you give them a different file name. Cura was never designed to handle the case where multiple profiles have the same ID. The ID is taken from the file name (without extension).

> @Ghostkeeper Is this inconsistent config directory behavior desired for different platforms?

I'm not entirely sure what you mean by inconsistent config directory behaviour. It's intended that Cura gets mounted on a temporary path on Linux because that's how AppImage works.
It's also intended that there are multiple directories on every platform where profiles get loaded from: Each platform can load profiles from their own installation as well as from a user-modifiable folder so that we can ship some profiles with Cura but the user can also install their own. This is precisely what you're doing now.
Lastly, it's also intended that the configuration directory is different on every platform. After all, Linux installations don't have an `%APPDATA%` folder and Windows installations don't have a `~/.local/share` folder. It must be different on every platform according to the common directory structure that they hold. Cura adheres to the standards in this regard.

> I could not find the documentation for .cfg files.

This is the file format: https://en.wikipedia.org/wiki/INI_file . See also Python's implementation of it [here](https://docs.python.org/3/library/configparser.html).