Ultimaker/Cura #5886: ooze shield's brim is merged with object's first layer

wuyuanyi135, 2019-06-12:
**Application Version**
4.1.0 Appimage

**Platform**
Ubuntu 18.10

**Printer**
N/A

**Actual Results**
When ooze shield is on, the brim of it will be merged with the object's first layer and unable to remove
[image]

Left: Ideamaker Right: Cura
[image]

**Expected results**
The brim should not be overlapping with the object

**Additional Information**
Project file
[CFFFP_cali_1-R.zip](https://github.com/Ultimaker/Cura/files/3279622/CFFFP_cali_1-R.zip)

rburema, 2019-06-12:
Still happens in master. I'll discuss it with the rest of the devs.

closed 2019-06-17 by Ghostkeeper

wuyuanyi135, 2019-06-20:
Thank you guys!

Ghostkeeper, 2019-06-21:
Should work in 4.2.