Ultimaker/Cura #5889: [bug] zoom out / zoom in

yet-another-average-joe, 2019-06-12:
**Application Version**
4.1

**Platform**
Windows 10

**Printer**
Does not depend on a printer

**Actual Results**
At maximum zoom out, the mouse wheel has no or very little effect zooming in/out

**Expected results**
Zoom should work...

**Workaround**
Clicking the perspective cube solves the problem until the next max zoom out

**Additional Information**

(searching with "zoom", I found no bug report...)

Liger0, 2019-06-12:
Same, I find hard to zoom in tiny details because, even if I have the "center visual on select object" option turned on and object selected, and also if not, I can't see the details of the model because it stops zooming in too early and at that point also zooming out is very hard.

Ghostkeeper, 2019-06-13:
I'm not able to reproduce this issue. At maximum zoom out, I indeed cannot zoom out further (so yeah the mouse wheel has no effect then). But I can zoom in fine. Scrolling upward will zoom in instantly and fast.

yet-another-average-joe, 2019-06-13:
This is weird. Yesterday, I had the problem on these 2 versions : official 4.1 and last Smartavionics branch.

Today, I can't reproduce the problem with official 4.1 (but it is still there with last Smartavionics branch). Therefore it disappeared after a reboot.

I was learning about json printer/extruder definition files and had many crashes (Cura does not like errors in extruder names !). I discovered this glitch, launched Smartavionics version, and it was also there. I will tinker again with these files until official Cura crashes, and see what happens. 

On my system, the 2 versions are easily recognizable, as  the plugin "Sidebar GUI" does not work with 4.1 official, and therefore has the orientation buttons in the bottom left corner. Furthermore, I did'nt have the black theme activated.

If I can reproduce the problem, I'll shoot a video.

yet-another-average-joe, 2019-06-13:
I found how to CREATE this bug !

1 - crash Cura so it will reset the user directory
2 - copy all the user's subdirectories from Smartavionics branch to the official 4.1 user directory

(it is the way I quickly recover from my failed experiments...)
It is stupid.

I close this "bug" report.

closed 2019-06-13 by yet-another-average-joe