Ultimaker/Cura #5890: [4.1] Adaptive Layer Varation step size  reset to 0 disabling second extruder and prevents slicing

Liger0, 2019-06-12:
**Application Version**
4.1

**Platform**
Win10 x64

**Printer**
Dual extruder custom printer

**Actual Results**
When disabling the second extruder and trying to slice, it gives an error because the Variation step size defaults to 0, and I have to set it manually to 0.04mm each time.

**Expected results**
Why the heck does it change that value? It shouldn't.

Ghostkeeper, 2019-06-13:
For me it defaults to 0.01mm and slices fine then. My reproduction steps were:

1. Add Custom FFF printer.
2. Set number of extruders to 2.
3. Disable the second extruder.
4. Load a cube. Let it slice.

When I enable the Adaptive Layer Height checkbox it says that the setting is 0.01mm (and still slices fine). Here is a project file: [CFFFP_cube.zip](https://github.com/Ultimaker/Cura/files/3284991/CFFFP_cube.zip)

Liger0, 2019-06-13:
I don't have the adaptive layer height and automatic layer height enabled. 
If I create a 2 extruders machine and set the adaptive layer height to 0.04, disable the adaptive layer setting, and disable the second extruder, the first extruder defaults to 0.00 adaptive layer height and slicing gives an error even tho the feature is disabled.

Liger0, 2019-06-13:
https://youtu.be/aShAdV3nAd8

Ghostkeeper, 2019-06-14:
Could you try the same reproduce steps, but with Custom FFF printer? The steps I'm trying according to your instructions are now:

1. Add Custom FFF printer.
2. Set number of extruders to 2.
3. Enable Use Adaptive Layers.
4. Set Adaptive Layers Variation Step Size to 0.04mm.
5. Disable Use Adaptive Layers.
6. Disable the second extruder.
7. Load a cube. Let it slice.

It slices fine for me. I believe you, but I'm not able to reproduce the problem on my computer yet :cry:

Liger0, 2019-06-14:
The printer I use is already a custom FFF one. :(

Ghostkeeper, 2019-06-17:
With your profile it's complaining about the prime tower position since the profile overrides that position but my custom FFF printer was not big enough to include your new position. Making the printer bigger in my machine settings fixed that and allows it to slice fine again. :confused:

Liger0, 2019-06-17:
Did you act exactly like in my video? Disabling same extruder and the options in that order etc.. :\

Ghostkeeper, 2019-06-18:
Yeah. Well, almost. I don't have the sidebar plug-in installed so I'm disabling the extruder via the panel for the printer configuration.

Perhaps you can share a project file? That could shed some light in where this value of "0" comes from.

Liger0, 2019-06-18:
> 
> 
> Yeah. Well, almost. I don't have the sidebar plug-in installed so I'm disabling the extruder via the panel for the printer configuration.
> 
> Perhaps you can share a project file? That could shed some light in where this value of "0" comes from.

Well.. as you see that happens without any model loaded.

Ghostkeeper, 2019-06-18:
Not happening here. Maybe it's been fixed in 4.2?

Ghostkeeper, 2019-06-18:
Ah, yeah, that was it. In 4.1 it does reproduce using that project file.

Ghostkeeper, 2019-06-18:
It seems that this commit fixed it: https://github.com/Ultimaker/Cura/commit/63cd3a52ce3f04a3c69a038ae158eba2ee406461#diff-10b03606c8df532a49117b75a8a32be3

closed 2019-06-18 by Ghostkeeper