Ultimaker/Cura #5891: [4.1]Slow material tab

Liger0, 2019-06-12:
**Application Version**
4.1

**Platform**
Win10 x64

**Printer**
-

**Actual Results**
Changing the different values make Cura slow for a set timeframe. 
What is weird is sometimes the words in the field disappear and then reappear later on, sometimes even after new words are inserted.

**Expected results**
It should work like before 4.0.

**Additional Information**
https://www.youtube.com/watch?v=LxzlqPhQ2e4

Ghostkeeper, 2019-06-13:
I can reproduce this, yeah. It's not necessarily freezing but it does sometimes temporarily revert the name change. Very strange.

Liger0, 2019-06-13:
When "Silver44" was written, I selected to delete 44 but it also deleted silver instead, which disappeared, appeared again, and disappeared again while it should have deleted 44.

rburema, 2019-06-14:
Hi @Liger, We've chosen to defer this for now, since it's mostly just a visual issue, and we're a bit understaffed at the moment.

fieldOfView, 2019-06-14:
Slightly off-topic, but is reworking the whole Preferences dialog to the Cura theme on the roadmap? I guess this would then be picked up.

Ghostkeeper, 2019-06-14:
Yeah it is. If the problem lies with the update mechanic of system-styled QML elements then it could be solved there.

fieldOfView, 2019-06-14:
> Yeah it is. 

When(ish)?

Ghostkeeper, 2019-06-14:
Good question. Cura is currently being overloaded with requests to add support for new projects within Ultimaker so I don't think it's going to happen for 4.2 or 4.3 :disappointed:

fieldOfView, 2019-06-14:
Oooh, "new projects"....

Ok, then it makes sense for me to try to fix some things in the current version.

closed 2023-06-13 by github-actions[bot]