Ultimaker/Cura #5893: Build plate adhesion problem

Vipcioo, 2019-06-13:
This is a example of generating an adhesion when I place another object next to the main one. Look at the settings in Cura 4.0 and look at the source file.
The rims should be around all objects. Now he does it only around one object.
After changing to 'Brim' he does the same, but the borders touch the round post, not the main element.

Printer - Kossel Delta Plus
Soft - Marlin 1.0.9, Cura 4.0

[image]
[image]

smartavionics, 2019-06-14:
Hello @Vipcioo , please save the project file (File -> Save) and then either rename the .3mf file to .zip or zip it up and attach the .zip file to this issue. Thanks.

Ghostkeeper, 2019-06-14:
Yes, I'd like to see that model too because my suspicion is that it's not level on the build plate or not flat on the bottom side.

Vipcioo, 2019-06-14:
Gentlemen, I apologize for the confusion. You are right. I checked again my project in Fusion 360 and founded a slightly moved 'offset plane', which resulted in this error.
This is how sometimes sitting at night over the project ends ;(

Thank's for the quick reply.

smartavionics, 2019-06-14:
OK, no problem, cheers!

closed 2019-06-17 by Ghostkeeper