Ultimaker/Cura #5894: [4.1.0] GUI unresponsive after changing settings

yuriy-skvortsov, 2019-06-14:
**Application Version**
4.1.0

**Platform**
Mac OS Mojave 10.14.5, MacBook Pro A1398, GeForce GT 750M / Intel Iris Pro

**Printer**
Custom (Tiertime Up mini 2)

**Actual Results**
After changing some options in Print settings (for example, checkboxes "Enable Acceleration Control", "Enable Jerk Control", especially when done multiple times, the GUI can freeze for about a minute (irrelevant of the model loaded).

**Expected results**
I'd expect the GUI to be responsive while changing the options.

**Additional Information**
Also problems with GUI responsiveness arise when scrolling the list of options to be enabled in preferences/settings window. Don't know, if it is the same bug or a different one. The list items just get displayed one by one, taking about 1-2 seconds per item. Selection and rotation of model does not work smoothly as well.

Here is what I believe is a relevant section of the log:
 
2019-06-14 12:37:24,808 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [136]: New error check scheduled.
2019-06-14 12:37:31,341 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._checkStack [140]: Need to check for errors again. Discard the current progress and reschedule a check.
2019-06-14 12:37:31,604 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [136]: New error check scheduled.
2019-06-14 12:37:35,380 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._checkStack [140]: Need to check for errors again. Discard the current progress and reschedule a check.
2019-06-14 12:37:35,665 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [136]: New error check scheduled.
2019-06-14 12:37:38,704 - DEBUG - [MainThread] cura.AutoSave._onTimeout [56]: Autosaving preferences, instances and profiles
2019-06-14 12:37:43,799 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._checkStack [140]: Need to check for errors again. Discard the current progress and reschedule a check.
2019-06-14 12:37:44,880 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [136]: New error check scheduled.
2019-06-14 12:37:48,532 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._checkStack [140]: Need to check for errors again. Discard the current progress and reschedule a check.
2019-06-14 12:37:48,931 - DEBUG - [MainThread] cura.Machines.MachineErrorChecker._rescheduleCheck [136]: New error check scheduled.
2019-06-14 12:37:56,778 - DEBUG - [MainThread] cura.AutoSave._onTimeout [56]: Autosaving preferences, instances and profiles
2019-06-14 12:38:28,128 - INFO - [MainThread] cura.Machines.MachineErrorChecker._setResult [188]: Error check finished, result = False, time = 39.2s

And here is the full log: https://www.dropbox.com/s/i2z9dgjhwi00ixe/cura.log?dl=0﻿

Ghostkeeper, 2019-06-17:
I'm unable to reproduce it, even on my fairly slow 5 year old laptop. I can click on Enable Acceleration and it'll take about 0.5 seconds for all settings to appear. If I click on it many times I can easily click faster than the refresh rate, and it puts the same things in the log as what you're reporting, but it still takes 0.5s after the last click to finally refresh.

Liger0, 2019-06-18:
To me on a R6 w/ vega 56 an 16gb ram dual channel it takes 15 to 20 seconds for dropdowns to open.

yuriy-skvortsov, 2019-06-18:
Ok, today I could not reproduce it (badly enough) anymore. Maybe just restarting Cura helped, I don't know. The worst delay was about ~4 seconds after clicking multiple times on "enable acceleration control". The loading of items in preferences was also ok.

Vandrasc, 2019-06-20:
Did you also have other applications open while the issue occurred? I also notice that the GUI can become slightly slower (but nothing close to 15-20 s) if Cura and other programs were open, simultaneously, for a while. Could you try restarting your machine, and run only Cura?

Liger0, 2019-06-20:
I only run cura even after a restart, that's always the same and the system is idle.

mastahype, 2019-09-10:
@yuriy-skvortsov is the entire GUI unresponsive, or just the visible fields? I'm trying to track down a similar defect where all edit boxes become disabled and you can no longer update any fields (as if the config went read-only).

yuriy-skvortsov, 2019-09-10:
Unfortunately I have not used Cura since that time much and the bug seems to be rare so can't reproduce. As I remember it, the whole thing was unresponsive.

closed 2024-12-07 by github-actions[bot]