Ultimaker/Cura #5895: Support brim

sfjuocekr, 2019-06-14:
**Application Version**
4.0.0

**Platform**
Windows 10

**Printer**
Ender-3

**Actual Results**
No brim on supports when build plate adhesion is set to none

**Expected results**
I expected a brim when I enabled "Enable Support Brim"

**Additional Information**
I have a rather large object which has more than enough surface to adhere to the build plate, the supports however are pretty sparse and I would like a brim on them.

If I enable brim under build plate adhesion, it works. But, I do not need a brim on the object itself.

Liger0, 2019-06-14:
I can't reproduce it, the support brim is there even tho the brim o nthe model is None, skirt or raft.

diegopradogesto, 2019-06-14:
I also can't reproduce it. Can you share a project file (File -> Save...)?

sfjuocekr, 2019-06-18:
I'll post those tomorrow when I get home

sfjuocekr, 2019-06-18:
No brims:

[image]
[image]
[image]

After enabling brim under build plate adhesion:

[image]

[CE3_DualMultiLetterBlocksV2.4-ILU-DAD-Z80.3mf.zip](https://github.com/Ultimaker/Cura/files/3303688/CE3_DualMultiLetterBlocksV2.4-ILU-DAD-Z80.3mf.zip)

sfjuocekr, 2019-06-18:
Maybe it only happens with tree supports?

Liger0, 2019-06-18:
Brim is inside the tree supports. :confused:

Ghostkeeper, 2019-06-20:
For all supports, the brim is printed towards the inside rather than towards the outside.

Ghostkeeper, 2019-06-20:
I can see from the screenshot that it is working :smiley:

sfjuocekr, 2019-06-25:
Aaaaah, sorry for the misunderstanding of support brim! It makes sense now!