Ultimaker/Cura #5896: material color icon or name when selecting material

Liger0, 2019-06-14:
**Is your feature request related to a problem? Please describe.**
When you select a material, only the name of the brand and the material type are shown, so if you have the same filaments, just in different colors, they are shown the same, and you don't know which one you selected, until the extruder tab is updated.

**Describe the solution you'd like**
It would be nice if there would be an icon (a round shape)? with a colour in.

**Describe alternatives you've considered**
You could state the colour name, though it would waste space, but it would still be even better then the colour in some situations if the colour are only slightly different.

diegopradogesto, 2019-06-14:
Do you mean something like this?

[image]

It's already there. If your material is defined with a color, then it will show the color. If it's defined as "Generic" color, then it shows only the default color for that material type.

Maybe I didn't understand correctly.

fieldOfView, 2019-06-14:
I think OP means in the dropdown menu for Material.

Ghostkeeper, 2019-06-17:
We're unable to style that drop-down menu at the moment. It's system-styled, and the system styling doesn't support changing any colours. We can basically only add unicode characters, disable/enable entries, add dividers and submenus.

The problem with this menu is that it needs submenus and can become quite long. The system-styled built-in menu can position the submenus overlaying the screen (possibly outside the Cura main window), so that it can become bigger than the Cura window. It will also decide on the position of the drop-down and sub-drop-down so that it doesn't go outside of your monitor.

We could implement all that, but it will become quite complicated to custom-style that. That's why we decided against that.

MalikDrako, 2019-12-20:
@Ghostkeeper has that changed at all in the past few months? I played with the QML and got something acceptable, at least on Windows, although it will need someone who is actually familiar with QML and how it's used in Cura to take it the rest of the way

MaterialMenu.qml
```
import QtQuick.Controls.Styles 1.4
// ...
// Insert into this section: "model: brandMaterials
//                delegate: Menu
//                {"
                    style: MenuStyle {
                        itemDelegate.label: Label {
                            color: styleData.selected || styleData.open ? "white" : "black"
                            text: /(.*)(#[a-fA-F0-9]{6})/.exec(styleData.text)[1]
                        }
                        itemDelegate.background: Rectangle {
                            color: styleData.selected || styleData.open ? "blue" : "white"
                            id: brandMaterialColorsBg

                            Rectangle {
                                color: /(.*)(#[a-fA-F0-9]{6})/.exec(styleData.text)[2]
                                //width: 18
                                //height: 18
                                width: UM.Theme.getSize("favorites_button_icon").width
                                height: UM.Theme.getSize("favorites_button_icon").height
                                border.width: UM.Theme.getSize("default_lining").width
                                border.color: "black"
                                anchors.verticalCenter: parent.verticalCenter
                                anchors.left: parent.left
                                anchors.leftMargin: 2
                            }
                        }
                        itemDelegate.checkmarkIndicator: Rectangle {
                            color: styleData.checked ? "black" : "transparent"
                            radius: 4
                            width: 8
                            height: 8
                            anchors.verticalCenter: parent.verticalCenter
                            anchors.left: parent.left
                            anchors.leftMargin: 1
                            anchors.topMargin: 1
                        }
                    }
					
// ...
// Insert into 
this section: "model: brandMaterialColors
//                        delegate: MenuItem
//                        {"
                            text: model.name + model.color_code
```
[image]

Ghostkeeper, 2019-12-23:
Nothing really changed here. The menu there is rather hard to customise since the submenus of materials need to align with the side of the monitor, sometimes extending in the other direction because there is no space on the user's screen. The OS normally does that but if we have to implement that it's a bit difficult.

Your solution works though, also on Linux, although the spacing is a bit different for me:

[image]

I didn't know that the item delegate could be customised with a rectangle background like that.

The actual implementation would need to be a bit neater. Instead of putting both the colour and the name in the item text and then parsing it out, it's better to just get both from the model.

Also Deception Purple is definitely one of my new favourite colours.

nallath, 2019-12-23:
I don't really like the way that this is fixed though. The main issue with it is the inability of controls 1.x to handle styling to the level that you would want it.

Controls 2.x has much better control for this, so we should probably just consider switching over to that instead.

Yian-Lai, 2019-12-30:
@Liger0 I am not sure if it's clear to you: you can also specify colors in the "Display name" of each material so it will show up in the menu. Maybe this will solve your problem?

Liger0, 2019-12-30:
Honestly I gave up on cura in the mid time because I've never had a successful print as it is the only slicer that causes major stringing no matter the retraction and settings I use while all other slicers succeed having the same settings.
Either way I wasn't very confident in putting in the name of the material because having much materials keeping all in order is confusing and the tones can be a little different.