Ultimaker/Cura #5897: code regresssion : Wrong temperature in Gcode with cura version 4.1.0

manuprendlair, 2019-06-14:
Hello,

I got an incorrect temperature in Gcode.
I have 260° in my print setting but gcode temperature is 275 °:

[image]

[image]

[adaptateur_helice.zip](https://github.com/Ultimaker/Cura/files/3291094/adaptateur_helice.zip)

It's look like to a code  regression.

Emmanuel FIANCETTE
From France

smartavionics, 2019-06-14:
> It's look like to a code regression.

Or could the temperature substitutions in the start gcode have some error? e.g. wrong keyword used, wrong brackets, etc.?

Ghostkeeper, 2019-06-17:
Could you submit a project file that has this problem? Under `File` -> `Save...`

I'm suspecting that it has something to do with the initial layer printing temperature. Cura doesn't just make up numbers by itself. Not usually, anyway.

closed 2020-06-19 by no-response[bot]