Ultimaker/Cura #5899: [4.1.0] "Print as support" does nothing

Kuchenmampf, 2019-06-14:
Precondition:
- A model was loaded into cura version 4.1.0

Action:
- Select model
- Add a support blocker
- Select "Per model settings" and change the mesh type of the blocker cube mesh to "print as support"
- Press "Slice" button

Expected:
- The blocker model is printed as support -> a cube shaped support structure is visible at the position of the support blocker model

Result:
- There is no effect. Nothing happens to the blocker mesh

[image]
[image]

smartavionics, 2019-06-14:
This is a known regression, I believe.

ultradyran, 2019-06-14:
experiencing same here

Ghostkeeper, 2019-06-17:
Duplicate of https://github.com/Ultimaker/Cura/issues/5813.

closed 2019-06-17 by Ghostkeeper