Ultimaker/Cura #5900: [4.1.0] Skirt or Brim does not scale when part is scaled

Beeblebrox666, 2019-06-15:
**Application Version**

4.1.0
**Platform**

Windows 10
**Printer**

Ender 3 Pro
**Actual Results**

Brim prints at the size as if the part was being printed at 100% Brim did not scale, also the supports on one end of the project piece printed inside the boundary of the part.
**Expected results**

The brim &/or supports need to scale with the part. In my example the part was being printed at 203% of original.
**Additional Information**

Beeblebrox666, 2019-06-15:
[Desktop.zip](https://github.com/Ultimaker/Cura/files/3293192/Desktop.zip)
This did not upload on the first attempt for somereason...

Ghostkeeper, 2019-06-17:
The brim and support do scale with the part. Even in your case. In your project file, the skirt becomes wider with 203% than it was at 100%.

The problem with your part is that the bottom is not flat on the build plate. The bottom part of your mesh does seem to be flat to me, but the mesh is not rotated properly to have that surface flush on the build plate. As a result, part of the bottom surface ends up on the second layer. If the model is scaled down to 100%, the difference in height is less than half a layer height so it happens to all end up on the first layer by chance.

If you press the "Lay Flat" button in the rotate tool, it solves the issue with your model.

closed 2019-06-17 by Ghostkeeper