Ultimaker/Cura #5901: Copied support blocker affected by "drop to build plate" option

chrullrich, 2019-06-16:
**Application Version**
4,1,0

**Platform**
Windows 10

**Steps to reproduce**

0. Ensure "automatically drop models to the build plate" is enabled
1. Add model
2. Add support blocker
3. Move support blocker above build plate; note that it stays where put
4. Select support blocker, multiply selected model
5. Move copied support blocker above build plate; note that it drops down

**Actual Results**
Copies of a support blocker created using "multiply selected model" cannot move above the build plate if the option to keep models on the build plate is enabled.

**Expected results**
Copies of support blockers behave like their originals and can freely move in Z direction regardless of preferences.

Ghostkeeper, 2019-06-17:
Yeah the copied support blocker is not linked to the model any more. Also if you move the printed model, only the original blocker is going to move along.

Maybe the copy should also be linked to that model?

Ghostkeeper, 2019-06-18:
We decided to defer this issue. Indeed the copy should also be linked to that model, but if you were to select both the model and the support blocker then you'd expect the copy of the support blocker to be linked to the copy of the model instead of to the original model. This then gets complicated real fast: What happens when you select 6 models and then copy: Two models with support blockers attached, one loose support blocker and one loose model. We don't think our time limits currently warrant planning a bugfix for this in.

closed 2023-06-24 by github-actions[bot]