Ultimaker/Cura #5902: Speed affected in new version

kornshadow097, 2019-06-16:
**Application Version**

4.1.0
**Platform**

Windows 10
**Printer**

tevo Tarantula 0.6mm nozzle
**Steps to Reproduce**

Just printing with this new version made anything made in fusion 360 print faster than previous version

updated from 4.0
**Actual Results**

everything prints faster and thus more chaotic
**Expected results**

for it to print the same speed i set in the profile i didnt change.
**Additional Information**

smartavionics, 2019-06-16:
Please save the project file (File -> Save), rename or zip it and attach the zip file to this issue. What speed are you expecting it to print at?

Ghostkeeper, 2019-06-17:
I'm not seeing any differences in the settings for Tevo Tarantula between 4.0 and 4.1 except the gantry height. The gantry height is not used in slicing.

Another difference is in the maximum resolution setting. That's not necessarily printing speed but will affect the total printing time.

closed 2020-06-19 by no-response[bot]