Ultimaker/Cura #5903: skirt feature will only put min of 4 skirts

kornshadow097, 2019-06-16:
title: skirt feature incorrectly working

**Application Version**

4.1.0
**Platform**

Windows 10
**Printer**

tevo Tarantula 0.6mm nozzle
**Steps to Reproduce**

put any number of skirts below the number 3.
**Actual Results**

it always puts 4minimum, even in the preview.
**Expected results**

it should put 1, 2 or 3... not 4+
**Additional Information**

smartavionics, 2019-06-16:
Hello @kornshadow097 , without the project file it's hard to be sure what the problem is but it's probably because the skirt/brim min length you are using is increasing the number of skirts.

[image]

kornshadow097, 2019-06-16:
Thanks. you were right. odd that it only started doing that in version 4.0+ I had that item hidden. maybe the default hidden values changed and screwed up my older profiles...

kornshadow097, 2019-06-16:
do you happen to know what items were changed from default after version 3.6.0? because ill have to go back and check every hidden setting and makes sure it wasnt automatically changed to something i DIDNT  want or setup. it might explain a few other features working strangely.

smartavionics, 2019-06-16:
Sorry, no I don't know what has changed. The changes could be in the base settings which all profiles inherit from or they could be in the printer specific settings.

kornshadow097, 2019-06-17:
so i just did a test print and it still acts weird. it used to automatically calculate what it would take to create a full circle of skirts i input that i wanted. I'd just leave the minimum off and it would not use that value which was default to 300mm which i guess is 4 skirts. 

Now it seems to go off the minimum, but it treats that minimum as an exact value. so if i put down 50mm it will print x skirts at an exact value of 50mm each. Even if 50mm isn't a full skirt, it will print whatever portion of the skirt is 50mm long and leave a gap after the last skirt.

instead of what its supposed to do which is use the minimum as a MINIMUM.  

Also, It no longer spaces the skirts apart, it lays them down next to each other like a brim which is fine but odd. Then it DOESN'T retract after the skirts are finished. It just move right into the project leaving a string of filament that leads into my project and connects it to the skirts...

Ghostkeeper, 2019-06-17:
> Now it seems to go off the minimum, but it treats that minimum as an exact value. so if i put down 50mm it will print x skirts at an exact value of 50mm each. Even if 50mm isn't a full skirt, it will print whatever portion of the skirt is 50mm long and leave a gap after the last skirt.

I'm unable to reproduce your bug that it prints half skirts. It always closes the contour completely for me. In the code, this setting is only used in the part that generates a polygon for the skirt or brim, and this polygon must always be closed so I don't see how it could ever do that. It doesn't check whether the minimum length is exceeded line by line.

> Also, It no longer spaces the skirts apart, it lays them down next to each other like a brim which is fine but odd.

The skirt has never spaced individual lines apart, unless you set the initial layer flow to be very low and the line width high (then it'll space all lines apart on the first layer).

Vandrasc, 2019-06-20:
Hi @kornshadow097, I am also not able to reproduce. It will help us a lot if you could share your project file with us.

Thank you!

closed 2020-06-19 by no-response[bot]