Ultimaker/Cura #5906: New material diameter not effective until after restart

larsch, 2019-06-17:
**Application version**: 4.0 + 4.1

**Platform**: Windows

**Printer**: CFDM

**Reproduction steps**: Create a custom material. Set diameter to 1.75 mm. Restart Cura. Duplicate previous material and change diameter of copy to something else, e.g. 1.9 mm. Select old material and slice. Select new material and slice.

**Actual results**: Both slicings result in same length of filament used (consistent with 1.75mm)

**Expected results**: The 1.9 mm material should use a shorter length of filament.

**Additional information**:

Original material:

[image]

Slice result:

[image]

Duplicated material & changed to 1.9mm:

[image]

Slice result after selecting new material:

[image]

Only after **restarting Cura** completely, it works as expected:

[image]

Ghostkeeper, 2019-06-18:
When trying to reproduce this bug, I found that it didn't update the diameter at all (under water). As in, it didn't affect the used filament length but also reset to the original diameter after restarting Cura. We may have a regression on top of this bug in 4.2 right now.

LipuFei, 2019-06-18:
JIRA ticket: CURA-6590

Ellecross, 2019-06-20:
Hey @larsch, FYI - the original issue you reported is now fixed in Cura master build. (yaay!) But we put the ticket back to do as there is still a remaining small issue to be resolved. So as what Ghostkeeper said, we expect this fix to be officially in 4.2. :D

closed 2019-07-02 by nallath

bullestock, 2020-05-13:
I can confirm that this bug is still present in Cura 4.5.0.

fieldOfView, 2020-05-13:
I ran into this issue with the Material Settings plugin, and the next version of that plugin will contain a fix for this issue. I will see if I can backport (forwardport?) this fix to upstream.