Ultimaker/Cura #5907: fan off with "print sequence" set to "one at the time"?

Christian72D, 2019-06-18:
It looks like the slicer does not turn off the fan AGAIN when the "print order" ist set to "one at the time", i have multiple objects and i tell cura to switch fan on after a certain layer height.

So the fan is off for the first object until the given height, then the fan turns on, and the first object will be printed.
Then it switches over to the next object but seems to forget to switch off the fan again.

The problem seems to occur with cura 4.0 and 4.1 on Windows.

Ghostkeeper, 2019-06-18:
Just to be clear, is this problem with the "Regular Fan Speed At Layer" setting, or with the Change At Z post-processing script?

Christian72D, 2019-06-20:
It´s with the "Regular Fan Speed At Layer", i normally don´t use pp scripts.

Ellecross, 2019-06-20:
Hey @Christian72D can you please send me your project file (including your machine + print settings)? 
This would be helpful to try and reproduce the issue. 

Thanks

closed 2020-06-19 by no-response[bot]