Ultimaker/Cura #5910: issue with the actiavtion "eviter les supports"

tarmoblue, 2019-06-18:
**Application version**

**Platform**

**Printer**

**Reproduction steps**

**Actual results**

**Expected results**

**Additional information**

tarmoblue, 2019-06-18:
quand on active décalage en Z lors d'une rétractation:
dans le Gode F est erronée:
sans décalage en Z lors d'une rétractation:
G0 X201.979 Y92.266
G1 F1500 E50.99777
;MESH:NONMESH
G0 F3600 X188.908 Y93.971
G0 X188.602 Y93.143
G0 X76.064 Y102.929
;TIME_ELAPSED:63.283563
;LAYER:1
G0 F5400 X76.064 Y102.929 Z0.4
;TYPE:SUPPORT
G1 F1500 E52.56014
G1 F2700 X77.034 Y102.986 E52.59633

avec décalage en Z lors d'une rétractation:
G1 X71.595 Y99.81 E36.6359
G1 F1500 E35.1359
G1 F17987547480000 Z0.7
G0 F3600 X73.287 Y101.571 Z0.7
;TYPE:SUPPORT
G1 F17987547480000 Z0.2
G1 F1500 E36.69827

closed 2019-06-18 by tarmoblue

Ghostkeeper, 2019-06-20:
I think you need to report this bug here: https://github.com/fieldOfView/Cura-ZOffsetPlugin/issues
And it helps if you use English. The internet doesn't speak French.