Ultimaker/Cura #5911: To log Temperature data of bed and nozzle

Adarshss03, 2019-06-19:
Our concern is to get the real-time temperature of the bed and nozzle. I don't know whether it is logged somewhere in the Cura directory for each project...

Ghostkeeper, 2019-06-20:
Cura doesn't log your bed/nozzle temperatures usually and never logs it. Cura is not your solution for this, right now.

Cura does query the temperatures every 2 seconds when connected via USB cable. Theoretically you could hook into this with a plug-in.

I don't think we're going to implement this logging. The Ultimaker printers do log their temperatures over time in the printer firmware, and I believe so does Octoprint.

pkuiper-ultimaker, 2019-06-20:
Hi,

You can see the real time log of an Ultimaker printer on the following this URL: 
http://**<YOUR PRINT IP ADDRESS>**/info/temperature.html

The data is on the machine itself (and has a limited buffer) and is not logged by Cura.

If you like to fetch the data yourself in JSON format, you can use the API directly:
http://<YOUR PRINT IP ADDRESS>/api/v1/printer/diagnostics/temperature_flow/20

The documentation for this API can be found here:
http://<YOUR PRINT IP ADDRESS>/docs/api/#!/Printer/get_printer_diagnostics_temperature_flow_sample_count