Ultimaker/Cura #5913: fail to slicing 3mf model file for cura 4.1

lyfbuaa, 2019-06-19:
fail to slicing 3mf model file for cura 4.1.
only one layer

smartavionics, 2019-06-19:
Hi @lyfbuaa , we are not mind readers, you are going to have to supply more information for your issue to be investigated. Please save the project file (File -> Save) and either rename to .zip or zip it up and attach to this issue. Thanks.

lyfbuaa, 2019-06-20:
pls investigate the attach
[CFFFP_16.zip](https://github.com/Ultimaker/Cura/files/3308378/CFFFP_16.zip)

when output the stl type,the slicing is normal

smartavionics, 2019-06-20:
Hello @lyfbuaa , thanks for the file. The problem appears to be caused by adaptive layers feature. If you disable that setting, it slices OK. I have checked the model and that looks OK so I guess this is a Cura bug related to adaptive layers.

smartavionics, 2019-06-20:
Hello again. I just discovered the reason it does not slice when using adaptive layers and that is because the model is not sitting on the buildplate. You need to set the z value to 0.

@fieldOfView , when I try to use the mesh tools on this model, they blow up:
```
2019-06-20 08:22:15,424 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [61]: An uncaught error has occurred!
2019-06-20 08:22:15,425 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]: Traceback (most recent call last):
2019-06-20 08:22:15,425 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "/opt/cura/lib/python3/dist-packages/UM/Qt/Bindings/ExtensionModel.py", line 51, in subMen
uTriggered
2019-06-20 08:22:15,425 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:     item["extension"].activateMenuItem(option_name)
2019-06-20 08:22:15,426 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "/opt/cura/lib/python3/dist-packages/UM/Extension.py", line 38, in activateMenuItem
2019-06-20 08:22:15,426 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:     self._menu_function_dict[name]()
2019-06-20 08:22:15,426 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "/home/markb/.local/share/cura/master/plugins/MeshTools/MeshTools/MeshTools.py", line 132,
 in checkMeshes
2019-06-20 08:22:15,426 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:     tri_node = self._toTriMesh(node.getMeshData())
2019-06-20 08:22:15,427 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:   File "/home/markb/.local/share/cura/master/plugins/MeshTools/MeshTools/MeshTools.py", line 234,
 in _toTriMesh
2019-06-20 08:22:15,427 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]:     indices = mesh_data.getIndices()
2019-06-20 08:22:15,427 - CRITICAL - [MainThread] cura.CrashHandler.__init__ [64]: AttributeError: 'NoneType' object has no attribute 'getIndices'
```

Also, when you select the model, it shows a weird frame around it. Where's that coming from?

[image]

diegopradogesto, 2019-06-20:
It does slice correctly to me (see image below).

@smartavionics , the "weird" frame around is because it's a grouped model. However if I ungroup it, I don't see any other model. @lyfbuaa can you explain why it's a grouped model. This is maybe the cause of the problem (I'm not sure)

[image]

diegopradogesto, 2019-06-20:
If it's related with adaptive layer heights, then it's maybe related with #3183 . If we can confirm this, then this ticket is a duplicate.

smartavionics, 2019-06-20:
Yes, it looks to be the same issue.

lyfbuaa, 2019-06-20:
> Hello @lyfbuaa , thanks for the file. The problem appears to be caused by adaptive layers feature. If you disable that setting, it slices OK. I have checked the model and that looks OK so I guess this is a Cura bug related to adaptive layers.

really,when uncheck the adaptive layers,slicing is normal

closed 2019-09-05 by diegopradogesto