Ultimaker/Cura #5914: Play sound efter slicing is finished.

SillyShitt, 2019-06-20:
**Is your feature request related to a problem? Please describe.**
 I often slice complicated files that take >30min to slice, and I don't want to sit by the computer waiting. 

**Describe the solution you'd like**
 A option to play a sound when slicing is finished would be AWESOME! 

**Describe alternatives you've considered**
 A system grawl or similar would be great also. 

**Affected users and/or printers**
 Anybody printing large print would be happier camper.  

**Additional context**
 With tree support even simple prints can take considerable time to slice. I could be doing the dishes instead of waiting.

Vandrasc, 2019-06-20:
Ha, that sounds nice, I will bring it up with the Cura team, we do not promise anything as we have other high priority implementation on the list. 
But we do happily accept a Pull request, for the technical ones :)

dstulken, 2019-07-12:
I'll second this as a good feature idea!  
OpenSCAD does this ("Play sound notification on render complete"), and it is incredibly handy.

fieldOfView, 2019-07-12:
This could easily be a plugin.

closed 2023-09-29 by github-actions[bot]