Ultimaker/Cura #5915: Enable support brim - shifts first layer. [4.1.0]-[4.0.0]

SillyShitt, 2019-06-20:
**Application version**
Cura 4.1.0 

**Platform**
Mac OS 

**Printer**
Custom Printer 

**Reproduction steps**
When using a "brim" for build plate adhesion and "concentric" support pattern, I then activate "Enable Support Brim" under Support, and all the support lines in the first layer gets shifted by the distance in "Support Brim Width ". 

**Actual results**
The second and all other layers of the support are now hanging in air, not connected to the first layer. 

**Expected results**
the first and second layer should be on top of one another, but they are now misaligned. 

**Additional information**
Seams to be reproducible with random objects when the object has several points connecting it to the build-plate and a large overhang. The extra brim to hold down support structures is a great idea, it just doesn't work with concentric support pattern.

Ellecross, 2019-06-20:
Hey @SillyShitt 

1. Do you see this shifting already in Preview > Layer View? 
2. I'm trying to reproduce it now, but I only have a UM printers. I'm printing out a model with suppport (enabled support brim + concentric pattern). I'll check if I see this too. 
3. In any case, can you send your project file (with your printer and settings) so that I can try and reproduce the problem? This will help us figure out the way to resolve this. 

Thanks!

Ghostkeeper, 2019-06-20:
This reproduces also with UMS5. Here is a 4.2 project file: [UMS5_gazebo.zip](https://github.com/Ultimaker/Cura/files/3309725/UMS5_gazebo.zip)

Vandrasc, 2019-06-21:
Devs see CURA-6606 for reference. 

@SillyShitt, we created a ticket and it is currently in our backlog. Thank you for your feedback!

LipuFei, 2019-08-23:
Fixed via https://github.com/Ultimaker/CuraEngine/pull/1117

closed 2019-08-23 by LipuFei