Ultimaker/Cura #5917: cura 4.1 Preview(Layer view) Skirt Line error

leey99, 2019-06-20:
**Application version**

4.1

**Platform**

Windows 10 64bit

**Printer**

Creality CR-10 S4

**Reproduction steps**

View: Preview(Layer view)
Build Plate Adhesion Type: Skirt
Skirt Line Count: 4
Skirt Distance: 5mm
Skirt Minimum Length: 250mm

**Actual results**

Skirt Line in Print result

**Expected results**

**Additional information**

[image]
[image]

diegopradogesto, 2019-06-20:
Can you please share the project file? (`File`->`Save...`)

smartavionics, 2019-06-21:
That just looks like some of the models are not flat on the buildplate.

leey99, 2019-06-24:
> Can you please share the project file? (`File`->`Save...`)

I uploaded the project file.
Please analyze it.

[Dental Demonstration Mandibule.zip](https://github.com/Ultimaker/Cura/files/3320296/Dental.Demonstration.Mandibule.zip)

smartavionics, 2019-06-24:
Hello @leey99 , as far as I can tell, it looks fine. You could increase the skirt distance to space it further from the first layer outlines like I show here.

[image]

BTW @diegopradogesto , when I sliced this, the UI came over all weird using 100% CPU on all the cores and the object list was animating! sort of scrolling up and down. This was without me moving the mouse or selecting anything. Never seen that before. Have a play with this project and see if it does the same for you!

Ghostkeeper, 2019-06-26:
> BTW @diegopradogesto , when I sliced this, the UI came over all weird using 100% CPU on all the cores and the object list was animating! sort of scrolling up and down. This was without me moving the [...]

Not seeing this happen. The model takes a while to load and this uses 1 core on 100% for about 5 seconds, but that's not out of the ordinary for loading a 300MB XML file. The object list only contains one object in the end and was not animating.

smartavionics, 2019-06-26:
If I remember right, after loading the model, I clicked on some of the teeth to check their z heights and it was after doing that things went weird.

Ghostkeeper, 2019-06-26:
The reason why the skirt looks so weird is because the models touch the build plate only in a very small spot in their centres. This makes the skirt go underneath the model, which makes it look like it's too small. But it's not. Here is what the first layer looks like:

[image]

The skirt goes around the first layer, not around all of the layers. That would not be useful. So this is working as intended as far as I can see.

closed 2019-06-26 by Ghostkeeper

smartavionics, 2019-06-26:
I also think that when it was being weird, the layer view was displayed not the prepare view.

leey99, 2019-06-26:
Thanks to you, I understand.
@smartavionics Thank You.
@Ghostkeeper Thank You.