Ultimaker/Cura #5918: Octoprint Button (direct acces after slicing) is Missing

nahuelcarou, 2019-06-20:
**Application version**

4 1 0 B
**Platform**

W10 . Other Data not relevant. 
**Printer**

RepRap
**Reproduction steps**

"Send to Octoprint Button"
**Actual results**

When it has sended to Octoprint, doesnt appearsthe button (direct access) to Octoprint. WIth the version 3 from Cura, was a Direct acces to the Octoprints Page. Was much easier. 
**Expected results**

The button with the direct acces to Ocoprint should be there.
**Additional information**

No needed. you know already what i am talking about....An thanks for fixing it :). Please dont be so burocratic...

Ghostkeeper, 2019-06-21:
If you've just upgraded from version 3, then you'll need to download the Octoprint plug-in again from the Marketplace.

Ghostkeeper, 2019-06-21:
I'm going to assume that was your problem. If not, you should let the author of that plug-in know, here: https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues

closed 2019-06-21 by Ghostkeeper