Ultimaker/Cura #5919: Strange Slicing Result

RocketNut, 2019-06-20:
**Application version**

**Platform**

**Printer**

**Reproduction steps**

**Actual results**

**Expected results**

**Additional information**

RocketNut, 2019-06-20:
I am getting some strange slicing result. The first screenshot shows what I am getting instead what it should look like (second screenshot).

The STL file is too large to attach here so here is a DropBox link::
[https://www.dropbox.com/s/wvijx0ikqe9e2qf/My%20Complete_Disk.stl?dl=0](url)

[image]

[image]

RocketNut, 2019-06-20:
I forgot to include a screenshot from my CAD that shows the disk as designed.
For some reason I can not pasted here!!!!!

smartavionics, 2019-06-20:
Now that's what I call a big model (note tiny build plate at the bottom)!

[image]

However, it has problems as loading it causes the Cura Mesh Tools plugin to say...

[image]

Fix the model and Cura should be able to slice it.

RocketNut, 2019-06-20:
Yes, it is very large. When I loaded it I scaled it down to 15%,  then rotated 90 deg to lay it flat. If it's not laid down flat you will have to use support. I design, (with 3D printing in mind) it with X-axis pointing upwards so no supports are needed.  

OH, maybe I should mention I am using 3,4,1, I have too many problems with 4.1.0.
Here is a screenshot of X-Ray view. I never used this features so I do not understand what it is telling me.

smartavionics, 2019-06-20:
I loaded your model into FreeCAD and got it to analyze it. Loads of problems as you can see. You need to fix your model before it can be sliced....

[image]

RocketNut, 2019-06-20:
OKI will turn over to CAD support to see what they say.
It's just really strange that it is ONLY this model that has problems.

Ghostkeeper, 2019-06-21:
Because RocketNut's screenshot didn't work (e-mail attachments to replies to Github don't automatically get posted), here's a screenshot of X-Ray View of your model in Cura:
[image]

Initially it showed nothing special but rotating the camera towards the side reveals a bunch of red areas.

> I never used this features so I do not understand what it is telling me.

X-Ray View will show the insides of your model. If there is extra geometry inside, or the walls are not watertight, then it will show red areas. This means that it is ambiguous what the inside of your model is.

Since your model has this extra geometry or missing geometry inside (depending how you look at it), it is ambiguous where the insides of your mesh are going to be. This is why Cura can't figure out what spaces to fill with skin, and why this specific model prints so strangely.

It is not something that we can really fix in Cura realistically. There are tools out there to repair your mesh that go by heuristics, like that the surface that has the fewest sharp corners is the one that you probably intended as the outside of your model. Some suggestions:

* FreeCAD (like SmartAvionics suggested)
* MeshLab
* Blender
* Netfabb

All of these work with heuristics, some better than others. I'd advise you to stay away from the voxel-based ones though, like MakePrintable. They tend to make your print very blocky which is disturbing when actually printed.

I'm closing it because it's not something we can realistically fix.

closed 2019-06-21 by Ghostkeeper

RocketNut, 2019-06-21:
OK
I can understand why there might be excess material in the model. It appears the functions I used to draw the part and followed this response to cad support.
Please close this support ticket.

Thanks for the great support.

RocketNut, 2019-06-21:
Bingo
I sent the message about excess material to my cad support. I did not do a drawing function correctly causing the excess material. When I redid the function everything fell into place and now slice up correctly, 

Thanks for the great help... Please close this support ticket. Have a great day. 

 
 
Closed #5919.