Ultimaker/Cura #5920: Multi-color with single extruder

mbartlett21, 2019-06-21:
**Is your feature request related to a problem? Please describe.**

I would like to be able to print multi-material models with only a single extruder, much like how the [Palette 2](https://www.mosaicmfg.com/products/palette-2) does it.
Currently, I generate a GCode with Cura and go through it with notpad, taking note of the lengths of each color of filament needed and feed them into the bowden tube.

**Describe the solution you'd like**

I would like Cura to give the lengths and color of filament to feed into the bowden tube.

**Describe alternatives you've considered**

I have been going through the generated gcode and getting values out of that, and I would like an easier method

**Affected users and/or printers**

People with single-extrude printers that want to print multi-material models.

**Additional context**

Model: 
https://cad.onshape.com/documents/e62c7c89cd1bbecc8a2d73f0/w/001c8f143683af185ca34884/e/345b4783b0e9e9d66485f896

Lengths:
70.45 red
24.95 yellow
25.22 red
24.92 yellow
Feed red in

Result: 
[image]

GCode:
```
;FLAVOR:Marlin
;TIME:221
;Filament used: 0.167874m
;Layer height: 0.3
;MINX:-26.57
;MINY:-25.57
;MINZ:0.3
;MAXX:24.57
;MAXY:25.57
;MAXZ:0.9
;Generated with Cura_SteamEngine 4.1.0
M140 S60
M105
M190 S60
M104 S200
M105
M109 S200
M82 ;absolute extrusion mode
G28 ;Home
G1 Z15.0 F6000 ;Move the platform down 15mm
;Prime the extruder
G92 E0
G1 F200 E3
G92 E0
G92 E0
G92 E0
G1 F2100 E-7.5
;LAYER_COUNT:3
;LAYER:0
M107
G0 F3600 X24.57 Y9.234 Z0.3
;TYPE:SKIRT
G1 F2100 E0
G1 F900 X24.569 Y9.427 E0.00963
G1 X24.56 Y10.154 E0.0459
G1 X24.506 Y11.27 E0.10164
[... 2653 lines cut ...]
M140 S0
;Retract the filament
G92 E1
G1 E-1 F300
G28 X0 Y0
M84
M82 ;absolute extrusion mode
M104 S0
;End of Gcode
;SETTING_3 {"global_quality": "[general]\\nversion = 4\\nname = Extra Fast #2\\n
;SETTING_3 definition = fdmprinter\\n\\n[metadata]\\ntype = quality_changes\\nse
;SETTING_3 tting_version = 7\\nquality_type = verydraft\\n\\n[values]\\nadhesion
;SETTING_3 _type = skirt\\nretraction_combing = noskin\\nspeed_slowdown_layers =
;SETTING_3  3\\nsupport_type = buildplate\\n\\n", "extruder_quality": ["[general
;SETTING_3 ]\\nversion = 4\\nname = Extra Fast #2\\ndefinition = fdmprinter\\n\\
;SETTING_3 n[metadata]\\nposition = 0\\ntype = quality_changes\\nsetting_version
;SETTING_3  = 7\\nquality_type = verydraft\\n\\n[values]\\ncool_min_layer_time =
;SETTING_3  2\\ninfill_sparse_density = 20\\nironing_only_highest_layer = True\\
;SETTING_3 nretraction_amount = 7.5\\nretraction_speed = 35\\nskirt_brim_minimal
;SETTING_3 _length = 300\\nskirt_gap = 15\\nskirt_line_count = 2\\nspeed_ironing
;SETTING_3  = 35\\nspeed_layer_0 = 15\\nspeed_print = 40\\nspeed_travel_layer_0 
;SETTING_3 = 60\\nspeed_wall_0 = 15\\nsupport_brim_enable = True\\nsupport_xy_di
;SETTING_3 stance = 0.4\\nxy_offset = -0.03\\n\\n"]}

```

Ghostkeeper, 2019-06-21:
I'm thinking this is the type of thing that a plug-in would be necessary for. How/where would the instructions get displayed? How would you indicate which part of your model gets extruded with which colour? It can get complex pretty quickly.

mbartlett21, 2019-06-21:
@Ghostkeeper I was thinking it could be an option for multiple extrusion

Ghostkeeper, 2019-06-26:
Well, but you can't change the settings per-extruder in Cura then, can you? At least, some of them can't change, like nozzle diameter, nozzle offset or temperature.

mbartlett21, 2019-06-26:
@Ghostkeeper 
I have installed the Chroma software from Mosaic Manufacturing and it does exactly what I want (it can give the lengths of filament needed for each bit)
Thanks anyway!

closed 2019-06-26 by mbartlett21