Ultimaker/Cura #5923: [4.1] unexplained retractions and coasting

kornshadow097, 2019-06-21:
**Application version**

4.1
**Platform**

**Printer**

**Reproduction steps**

i printed a circular cap with eccentric infill. coasting off,z pattern at sharpest edge
**Actual results**

it normally had no reason to retract but eccentric infill somehow made it retract on many inner and outer walls. then on the outer most 2 walls it would print 1 wall and literally coast the entire outer wall...every layer. i made this project before and literally only changed infill from lines to concentric
**Expected results**

well i had a horrible print that took much longer to finish
**Additional information**

smartavionics, 2019-06-21:
Yeah, well, unless you provide the horrible gcode file and also the Cura project file (.3mf) there's not a chance that this is going to receive any attention.

Ellecross, 2019-06-24:
Hey @kornshadow097, like what SmartAvionics said, can you provide us the project file, please? file > save 
I'd like to see your settings so that I can understand how we can reproduce this.

closed 2020-06-19 by no-response[bot]