Ultimaker/Cura #5924: [4.0.0] Pause at Height plugin restarts too high

bluegizmo83, 2019-06-21:
**Application version**
4.0.0

**Platform**
Windows 10 X64

**Printer**
Ender 3 Pro

**Reproduction steps**
Enable the Pause at Height plugin, select Layer to pause at (12 in my case), leave redo layers at 0, slice and print.

**Actual results**
The printer pauses at the layer as expected (at the end of layer 12 / start of layer 13), print head raises and moves to the set position, and upon restarting the Z height is up too high when it starts printing the first layer after pause. It looks to be about one or two layers too high. successive layers are ok after that though, just that first layer after the pause is too high, almost like it skips a layer or two when it goes back to printing.

**Expected results**
Should start printing at the correct layer height when resumed.

**Additional information**

pkuiper-ultimaker, 2019-06-24:
I tested the LayerAtHeight script on Cura 4.1 with an Ultimaker 2 (has Marlin firmware like your machine)

It produced this G code ( added some comments in code to explain):

;TYPE:CUSTOM
;added code by post processing
;script: PauseAtHeight.py
;current layer: 12
M83
G1 F300 Z2.92                                   # The head move 1mm above its current z position
G1 F9000 X190 Y190                         # Head moves to the side
G1 F300 Z15                                      # Below 15mm the head is moved even higer to 15mm
M0;Do the actual pause                   # Pause the machine
G1 F300 Z2.92                                  # Continue after pause, move 1mm above last layer position
G1 F9000 X107.164 Y107.164          # Move back to old XY position
G1 F9000
M82
G92 E272.4471
;LAYER:12                                        # Next layer starts
;MESH:___AA-Cubet.stl
G0 X107.164 Y107.164 Z2.07          # Head moves to correct z height 
                                                        # (1,92mm old z pos. + 0,15mm  layer height = 2,07 new z pos.)

The script seem to work OK. Are you sure that the printer is printing a FULL layer at the incorrect height or is you printer oozing material at the return position (which is 1mm above the last z position) before it starts printing layer 13 (which bring it back to the correct z height)?

Please check the G code and see if the error is in the G code or in the printer (Oozing).

Ghostkeeper, 2019-06-26:
The bug reproduces for me. Here's a snippet of g-code that I get:

```
G0 F600 X145.882 Y154 Z2.6 ;<<<<<last height: 2.6mm
G0 F7200 X145.759 Y145.759
;TIME_ELAPSED:117.972096
;TYPE:CUSTOM
;added code by post processing
;script: PauseAtHeight.py
;current layer: 12
M83
G1 F300 Z3.6 ;<<<<<<<< post-processor moves up to 3.6mm
G1 F9000 X190 Y190
G1 F300 Z15
M104 S0; standby temperature
M0;Do the actual pause
M109 S200; resume temperature
G1 F300 Z3.6 ;<<<<<<<<<<<<<<< again moving to 3.6mm?
G1 F9000 X8 Y8
G1 F9000
M82
G92 E66.22521
;LAYER:12
M204 S500
M205 X8 Y8
;TYPE:FILL
;MESH:cube.stl
G1 F1200 X154.239 Y154.239 E66.62408 ;<<<<<< starts printing infill at 3.6mm!
```

The Z hop height of Creality CR-10 is 1mm. Perhaps that has something to do with it?

tlhintoq, 2019-07-22:
Here's another example.  Examined the gCode produced by 4.2 beta - Oddly the same model on the same settings doesn't do this for me with 4.0.0 - but now that I know it does for other's I'll be making a point of hand examining & editing every 2 color job to be safe.

Pretty clear that what's happening is a rise of exactly 1mm, then a second rise to clear any collisions before moving to the designated "resting point" set in the extension.  Then when the head returns it returns to the point 1mm high. _THEN_ on the _next_ layer tries to **drop** to the correct height.  Makes real mess.

[image]

mrVanboy, 2019-08-05:
Looks like this line has a bug:
https://github.com/Ultimaker/Cura/blob/8179e1e/plugins/PostProcessingPlugin/scripts/PauseAtHeight.py#L312
It should return nozzle to the original `Z` position, but it returns to `Z+1` instead.

pkuiper-ultimaker, 2019-08-05:
This is by design. You do not want to bump into the object when moving back after a pause. If you look at the generated g code you will see that the next layer start after this code and brings it back to the correct layer height.

tlhintoq, 2019-08-05:
@pkuiper-ultimaker 
No.  There is a bump up of 1mm
Then a bump up of 15
Then a return to the first bump up - 1mm too high.
And it starts printing there: Too high.
Then the _next_ layer is is a drop to {original layer}+layer height

It happens every time in 4.0, 4.1 and now 4.2.1 when using the pause at height add-on.
My normal routine now is to hand edit the generated gCode in notepad++.
I search for "pause" see the bump up line, and just drop the second line by 1mm.
Then I get a perfect change at layer height.
If you like I'll slice a benchy for ya and up load it so you can see for yourself.  Or I can slice one and upload screen shots of the relevant code segments.  Your call.

dehumbertman, 2019-08-07:
Thanks. Can confirm that manually editing the gcode like this works.  Now its just up to Ultimaker to get rid of this problem.  I also learned a valuable lesson.... never upgrade software before i have an ordered part to print.

tlhintoq, 2019-08-07:
> never upgrade software before i have an ordered part to print.

I have Versions 15.04 - 3.61 - 4.0 - 4.21 installed all side by side.  Just in case.

sitalkio, 2019-08-18:
I had the exact same problem in Cura 4.2.1yesterday.
Solved by manually editing the gCode just like @tlhintoq suggested but this really needs to be fixed.

gbushta, 2019-08-29:
Thanks for working on this and suggesting how to fix it. I'm using version 4.2.1 and after the pause on Ender 3 the height went all the way to the bottom and started printing.  It rammed into the existing print. Since I paused at layer 17 I am thinking of printing again with a negative number in the 'redo layer' section.  I'll see if redo layer -17 works.  Otherwise I'll just edit the gcode, or down grade to 4.0.
This is the first time I have use the Pause at Height so I can change to a different filament color.

tlhintoq, 2019-08-29:
Layer numbers are absolute not relative.  -17 makes no sense.  If you want it to resume at 17 just use 17.  Its not relative to current position such as being at 34 so -17 would take to positive 17: that's not how it works.

> after the pause on Ender 3 the height went all the way to the bottom and started printing.
I've used the pause at height with specified layers a LOT and never had it do that.  I'm not a Cura worker but I'd be interested to see your actual settings for that did this (screen shot) - and maybe the gcode that was generated.  You can't attach .zip or gCode to threads here: Just images.  If you wanted to rename the file  `blahblah.gcode.jpg` so it will upload and attach that might work.

DragNfLy, 2019-08-31:
@tlhintoq 
I'm also experiencing the 1mm too high start after pause-at-height.
Please help me change this code so my print starts again at the correct spot. I've tried a few different settings but not quite sure which to change. 

Much appreciated.
[code]
G1 F1800 X124.632 Y106.845 E1000.7468
G1 F3000 E994.7468
;MESH:NONMESH
G0 F600 X124.632 Y106.845 Z1.64
G0 F9000 X126.246 Y107.905
G0 X126.295 Y108.693
G0 X130.457 Y112.877
G0 X147.64 Y115.25
G0 X149.244 Y118.362
G0 X155.517 Y124.321
G0 X166.987 Y124.836
G0 X166.995 Y125.78
;TIME_ELAPSED:1125.528223
;TYPE:CUSTOM
;added code by post processing
;script: PauseAtHeight.py
;current layer: 6
M83
G1 F300 Z2.6399999999999997
G1 F9000 X190 Y190
G1 F300 Z15
M104 S195; standby temperature
M0;Do the actual pause
M109 S205; resume temperature
G1 F199.998 E100
G1 F300 Z2.6399999999999997
G1 F9000 X133.004 Y125.78
G1 F9000
M82
G92 E940.51488
;LAYER:5
G1 F3000 E940.51488
;TYPE:WALL-OUTER
[/code]

bkp23, 2019-09-03:
The problem is that PauseAtHeight.py returns back to the "1mm higher" height in line 312, then returns the X-Y back to the pause point, but never moves back down to the starting height.

In the first example above, we can see where the code _after_ the script happens to return to a known Z position. This does not always occurs and the script should not depend on this occurring. It's causing many resumes to occur 1mm too high for a single layer.

tlhintoq, 2019-09-03:
@bkp23 
Terrific news! Thank you.
So... I'm not well versed on python scripting. But am I correct in thinking that there is a script file for this installed on the client machine (our PC's) - and that we should be able to replace it with your latest version and be good-to-go? IE: No need to wait for a new version of Cura-We just need to replace the script?

DirtyRat91, 2019-09-04:
@DragNfLy , in your gcode you'll see just below the 

;MESH:NONMESH
G0 F600 X124.632 Y106.845 Z1.64

Mesh line is the last time your printer adjusts the Z height to 1.64. 

Then when the pause at height script modifies the code to:

;script: PauseAtHeight.py
;current layer: 6
M83
G1 F300 Z2.6399999999999997
G1 F9000 X190 Y190
G1 F300 Z15

It raises it to 2.64mm, then to 15mm.

Then right after the pause:

M109 S205; resume temperature
G1 F199.998 E100
G1 F300 Z2.6399999999999997

It SHOULD set the Z height back to Z1.64, however the bug has it 1mm higher. Simply change this line to 

G1 F300 Z1.64

Then your print should work fine. 

This is a tip that I've found useful to me. On my Ender3, the extruder is attached to the Z gantry. When changing filament I believe I've bumped the gantry down a pinch before. As the printer only knows its height relative to how much it has driven the stepper motor, it does not realize the Z height is lower due to my manually changing filament and bumping the Z gantry down. This caused a print to fail as it tried to print the next layer at approximately the same height as the previous, causing smudging and under extrusion. 

So in order to make sure that my printer is returning to the proper Z height, I rehome the Z axis, then run the final Z height command before it returns to printing. This is easily done with a 

G28 Z 

command right before your G1 F300 Z1.64 line.

WARNING, this does mean that the nozzle will lower to the print bed before rising back up to the Z height. Assuming you're away from the print at your parked X, Y position of 190, 190 that shouldn't cause a problem, make sure you're nowhere near your print.

tlhintoq, 2019-09-04:
@DragNfLy @DirtyRat91 
You may have noticed that @bkp23 has submitted a [fixed script](https://github.com/Ultimaker/Cura/pull/6298).

It still has to be approved and merged in but he did provide a direct link and I've tested it.  As you can see in that linked conversation I've tried it and it does seem to fix the problem.

For convenience I've attached the script here.  This site only allows the attaching of images so I've added a .jpg extension to the script.  After you've downloaded the file remove the .jpg extension.  Replace the script in your scripts directory.
`C:\Program Files\Ultimaker Cura 4.2.1\plugins\PostProcessingPlugin\scripts`
In the case of version 4.2.1 - you may have to adjust if you did a custom location or have an older version of Cura.
[image]

diegopradogesto, 2019-09-04:
The PR has been merged, so I'm closing this ticket.

Thank you all for the report, the investigation, the testing and the support.

closed 2019-09-04 by diegopradogesto