Ultimaker/Cura #5925: Initial Layer Flow affects whole print

john2288, 2019-06-21:
**Application version**
4.1.0

**Platform**
Windows 10 home

**Reproduction steps**
slice a print at 100% initial layer flow and 100% flow
Slice a print at 120% initial layer flow and 100% flow
compare gcode

**Actual results**
the 120% initial layer flow file also increases the other layers by 120%

**Expected results**
only first layer should print at 120%

**Additional information**
my printer likes a higher initial flow for adhesion, however it casues the other layers to squish by not returning to the flow setting.
100% initial layer
[image]

120% initial layer
[image]

As seen above the initial layer does the some moves but the extrusion is increased as expected by 120%

100% other layer flow
[image]

120% other layer flow
[image]

As seen above for example on line 6247, the move itself is the same however the extrusion values are not the same and the extrusion is increased by 120% on the second image

smartavionics, 2019-06-21:
Hello @john2288, please save the project file (File -> Save) and either rename to .zip or zip it and attach to this issue.  Without that it's not possible to investigate. Thanks.

john2288, 2019-06-21:
Sorry about that. i saved it but forgot to attach it.
[a flow test.3mf.zip](https://github.com/Ultimaker/Cura/files/3315650/a.flow.test.3mf.zip)

smartavionics, 2019-06-21:
Thanks for the file. I think Cura's OK and you are getting misled by the fact that the first layer is lower than the others. That means that the extrusion rate for the first layer is proportionally lower. So if I set the first layer height to be 0.4 (same as the others) the extrusion rate used for the wall lines is 0.08 mm/mm and for subsequent layers it's 0.067 which is a ratio of 1.19 (or thereabouts). Hope this helps.

john2288, 2019-06-21:
thanks for your reply.
I understand what you were saying but still wasn't getting it. however as i was trying to put together an image to maybe explain better i realized how what you said makes sense and completely forgot about the fact that the e value is absolute not relative and constantly goes up so the initial 120% will make the other values inflated.
My apologies for the trouble.

closed 2019-06-21 by john2288