Ultimaker/Cura #5926: Can't change active material using menu bar

giacomociani, 2019-06-21:
**Application version**
Cura 4.1.0

**Platform**
Windows 10 Enterprise

**Printer**
Creality Ender 3

**Reproduction steps**
- Open Cura, apply "normal" profile.
- Go to the menu bar -> settings -> Extruder 1 -> material -> and change change material (say between PLA and ABS). Material-related settings (e.g. printing temperature) do not change.
- Go to menu bar -> settings -> Extruder 1 -> material and notice that the dot indicating the selected material has not changed.
- Go in "prepare" environment, and change material (say between PLA and ABS) from the large information bar just under the label "prepare". Material-related settings (e.g. printing temperature) change according to your choice, and the dot in the drop-down menu from the menu bar has moved.
- also changing material from ctrl-K dialog using the "activate" button works correctly.

**Actual results**
Changin material from the information bar in prepare mode, or from the material setting dialog works correctly. Changin it from the menu bar drop-down list has no effect.

**Expected results**
Is should be possible to change material from menu bar drop-down list, or this list should be removed.

diegopradogesto, 2019-06-23:
I can't reproduce the issue. It works for me with any printer. Could you provide your 4.1 configuration folder (Go to `Help` > `Show configuration folder`)

Ghostkeeper, 2019-06-26:
Indeed this reproduces for me. My configuration folder minus log file: [4.1.zip](https://github.com/Ultimaker/Cura/files/3328755/4.1.zip)

giacomociani, 2019-06-26:
Here is my config folder too.
[4.1.zip](https://github.com/Ultimaker/Cura/files/3330452/4.1.zip)

LipuFei, 2019-08-27:
I cannot reproduce this now.

Ellecross, 2019-08-29:
I cannot reproduce this anymore either. 
@giacomociani can you please try on the latest Cura version? 
For now, we will close this.

closed 2019-08-29 by Ellecross

giacomociani, 2019-08-29:
This still reproduces for me on Cura 4.2.1.

Let me know if I can provide any other info.

Giacomo

LipuFei, 2019-09-02:
Hi @giacomociani , I can confirm that this is reproducible with Cura 4.2.1. This is fixed in the current `master` branch, which will be included in the next release.

Ghostkeeper, 2019-09-02:
Nallath fixed this together with a report that the nozzle couldn't be changed in the menu bar.

giacomociani, 2019-09-03:
Great! Thanks you.

Giacomo