Ultimaker/Cura #5927: [4.1] Wrong PREVIEW

leonirz, 2019-06-22:
**Application version**

4.1

**Platform**

WINDOWS 10

**Printer**

Creality cr-10

**Reproduction steps**

1) Open STL
2) Slice
3) PREPARE mode is all right.
4) Preview mode (where presents much errors in view)

**Actual results**

In PREVIEW mode, areas that should be open (recessed) are closed (filled) as a single plane. If you compare the PREPARE mode with the PREVIEW mode, you may notice the error. So as you navigate through the layers, the first layers are mostly with many errors, because areas that should appear open (cropped) are filled.

**Expected results**

View of piece right as PREPARE mode.

**Additional information**

[image]
[image]
[image]
[image]
[LOG cura.zip](https://github.com/Ultimaker/Cura/files/3316822/LOG.cura.zip)
[CCR10_CAIXA-FRENTE-V4.zip](https://github.com/Ultimaker/Cura/files/3316823/CCR10_CAIXA-FRENTE-V4.zip)

smartavionics, 2019-06-22:
Hello @leonirz , your model has problems as detected by the mesh tools and the x-ray view...

[image]

[image]

Red regions have problems.

leonirz, 2019-06-22:
I have no idea what these red notes mean. Nothing appears in the design program. I imagine it will be necessary to draw everything again from zero. :(

leonirz, 2019-06-22:
My CURA program does not have this MESH tool.

smartavionics, 2019-06-22:
> My CURA program does not have this MESH tool.

It's a plugin you install from the Cura marketplace...

[image]

diegopradogesto, 2019-06-23:
The problem is related with the model. Please fix the model and see if the problem stays.

It's a good idea to install the MeshTool plugin for that purpose, as @smartavionics suggested.

closed 2019-06-23 by diegopradogesto