Ultimaker/Cura #5929: [Feature Request]: Escape should cancel drag

lukehutch, 2019-06-23:
When dragging an object in Cura (to translate/rotate etc.), Escape should cancel the drag. However, this does not currently work (in Linux at least).

This is particularly important because it's easy to accidentally move an object when you mean to rotate the view, given that different 3D programs assign different buttons to rotate. When you move an object, it destroys the current slicing, and you have to slice the object again.

nallath, 2019-06-24:
Wouldn't it be better to be able to lock the object a better solution? Canceling a tool operation (and keeping the sliced object) is probably going to be a bit tricky, as we already apply the "in between" moves.

lukehutch, 2019-06-24:
What about just remembering the start position/orientation when a drag is initiated, and if Escape is pressed, move back to the original position/orientation?

Locking is not a general fix, because I'm talking about avoiding inadvertent drags.

nallath, 2019-06-24:
Well, that's what the ctrl + z does, but at that point, we already deleted the stored slice data.

GregValiant, 2024-11-25:
I'm cleaning house.
Is this still a relevant request or can it be closed?

GregValiant, 2024-11-25:
The ESC key used to toggle the Cura app window from "Normal" to "Maximized".  In the current version (5.9.0) I'm not sure what it does but it no longer affects the app Window.
I'll leave this open.