Ultimaker/Cura #5930: RuntimeError: the sip module implements API v12.0 to v12.4 but the Arcus module requires API v12.6

francoisbrassart, 2019-06-23:
**Application version**
master branch

**Platform**
Ubuntu 18.04

**Reproduction steps**
I followed this steps exactly : [https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source-on-Ubuntu](url).I installed all the libraries in /usr/lib (so I had to be super user to do it)

**Actual results**
```
titi@ubuntu:~/Documents/Cura$ python3 cura_app.py
/home/titi/Documents/Uranium/UM/PluginRegistry.py:4: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
Error in sys.excepthook:
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/titi/Documents/Cura/cura/CuraApplication.py", line 27, in <module>
    from UM.Qt.QtApplication import QtApplication  # The class we're inheriting from.
  File "/home/titi/Documents/Uranium/UM/Qt/QtApplication.py", line 16, in <module>
    from UM.Backend.Backend import Backend #For typing.
  File "/home/titi/Documents/Uranium/UM/Backend/Backend.py", line 12, in <module>
    from UM.Backend.SignalSocket import SignalSocket
  File "/home/titi/Documents/Uranium/UM/Backend/SignalSocket.py", line 4, in <module>
    import Arcus
RuntimeError: the sip module implements API v12.0 to v12.4 but the Arcus module requires API v12.6

Original exception was:
Traceback (most recent call last):
  File "cura_app.py", line 132, in <module>
    import Arcus #@UnusedImport
RuntimeError: the sip module implements API v12.0 to v12.4 but the Arcus module requires API v12.6
titi@ubuntu:~/Documents/Cura$ sudo python3 cura_app.py
Traceback (most recent call last):
  File "cura_app.py", line 11, in <module>
    from UM.Platform import Platform
ModuleNotFoundError: No module named 'UM'
```

I saw issue #5683 but when I try purge sip, I got
```
sudo apt-get purge python3-sip
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package 'python3-sip' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 18 not upgraded.
```

pkuiper-ultimaker, 2019-06-24:
Hi,

Can you check your sip version?

```> python3
Python 3.7.1 (v3.7.1:260ec2c36a, Oct 20 2018, 14:57:15) [MSC v.1915 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import sip
>>> sip.SIP_VERSION_STR
'4.19.8'
>>> sip.__file__
'C:\\workspace\\installations\\python\\3.7.1\\lib\\site-packages\\sip.pyd'
>>>```

francoisbrassart, 2019-06-24:
Hi,

```
>>> import sip
>>> sip.SIP_VERSION_STR
'4.19.8'
>>> sip.__file__
'/usr/local/lib/python3.6/dist-packages/sip.so'
```

LipuFei, 2019-06-24:
@francoisbrassart what if you run `import Arcus` in that same console you run `import sip` + `sip.__file__`? Do you get the same error?

francoisbrassart, 2019-06-24:
Yes, same error...
```
>>> import sip
>>> sip.__file__
'/usr/local/lib/python3.6/dist-packages/sip.so'
>>> import Arcus
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: the sip module implements API v12.0 to v12.4 but the Arcus module requires API v12.6
```

pkuiper-ultimaker, 2019-06-24:
OK. It seems that libArcus might have been compiled against a different python version.

Can you rebuild libArcus (please first delete the CMake cache) and carefully check the CMake output. 
Make sure it finds the correct python path (/usr/local/lib/python3.6/ in your case).

francoisbrassart, 2019-06-24:
```
titi@ubuntu:/usr/lib/libArcus/build$ sudo rm CMakeCache.txt 
[sudo] password for titi:
titi@ubuntu:/usr/lib/libArcus/build$ sudo cmake ..
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found Protobuf: /usr/local/lib/libprotobuf.a;-lpthread (found suitable version "3.5.1", minimum required is "3.0.0") 
 -- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.8", minimum required is "3.4")
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found suitable version "3.6.8", minimum required is "3.4") 
-- Found SIP: /usr/bin/sip (found version "4.19.8") 
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_VISIBILITY - Success
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY
-- Performing Test COMPILER_HAS_HIDDEN_INLINE_VISIBILITY - Success
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR
-- Performing Test COMPILER_HAS_DEPRECATED_ATTR - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /usr/lib/libArcus/build
```
Looks like it found the good path for Python

francoisbrassart, 2019-06-24:
Ok maybe I did something wrong, now I don't have this error anymore but the 2nd one
```
titi@ubuntu:~/Documents/Cura$ python3 cura_app.py 
Traceback (most recent call last):
  File "cura_app.py", line 11, in <module>
    from UM.Platform import Platform
ModuleNotFoundError: No module named 'UM'
titi@ubuntu:~/Documents/Cura$ sudo python3 cura_app.py 
Traceback (most recent call last):
  File "cura_app.py", line 11, in <module>
    from UM.Platform import Platform
ModuleNotFoundError: No module named 'UM'

```

LipuFei, 2019-06-24:
The `sip` executable it's using doesn't seem correct:
```
-- Found SIP: /usr/bin/sip (found version "4.19.8") 
```
Your `sip` shown from the python console comes from `/usr/local` but this one comes from `/usr`. Can you check if you have `/usr/local/bin/sip`?

pkuiper-ultimaker, 2019-06-24:
The 'UM' module is from Uranium. Did you add the Uranium path to env?
This instruction:

$ export PYTHONPATH=[URANIUM_PATH]:${PYTHONPATH}

francoisbrassart, 2019-06-24:
@pkuiper-ultimaker My bad, when I export this path, I'm going back to the initial error

```
titi@ubuntu:~/Documents/Cura$ export PYTHONPATH=~/Documents/Uranium:${PYTHONPATH}
titi@ubuntu:~/Documents/Cura$ echo ${PYTHONPATH}
/home/titi/Documents/Uranium:
titi@ubuntu:~/Documents/Cura$ python3 cura_app.py 
/home/titi/Documents/Uranium/UM/PluginRegistry.py:4: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
Error in sys.excepthook:
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/titi/Documents/Cura/cura/CuraApplication.py", line 31, in <module>
    from UM.Qt.QtApplication import QtApplication #The class we're inheriting from.
  File "/home/titi/Documents/Uranium/UM/Qt/QtApplication.py", line 16, in <module>
    from UM.Backend.Backend import Backend #For typing.
  File "/home/titi/Documents/Uranium/UM/Backend/Backend.py", line 12, in <module>
    from UM.Backend.SignalSocket import SignalSocket
  File "/home/titi/Documents/Uranium/UM/Backend/SignalSocket.py", line 4, in <module>
    import Arcus
RuntimeError: the sip module implements API v12.0 to v12.4 but the Arcus module requires API v12.6

Original exception was:
Traceback (most recent call last):
  File "cura_app.py", line 127, in <module>
    import Arcus #@UnusedImport
RuntimeError: the sip module implements API v12.0 to v12.4 but the Arcus module requires API v12.6

```
@LipuFei Nope, Don't have /usr/local/bin/sip...
```
titi@ubuntu:~/Documents/Cura$ ls /usr/local/bin/
CuraEngine  miniterm.py  miniterm.pyc  protoc  __pycache__  pylupdate5  pyrcc5  pyuic5
```

pkuiper-ultimaker, 2019-06-24:
OK. Then now run "cmake-gui" and point to the libArcus build folder. 
Force Cmake to find SIP in '/usr/local/lib/python3.6/dist-packages/sip.so'  not in '/usr/bin'.
Configure, generate and build again. Check the CMake output to see if the correct sip executable is found,

francoisbrassart, 2019-06-24:
I did what you said : in cmake-gui, I linked the SIP EXECUTABLE to the value "/usr/local/lib/python3.6/dist-packages/sip.so".

When I go back in the build folder of libArcus, I have this error 
```
sudo make
[ 27%] Built target Arcus
[ 31%] Generating python/sipArcuspart0.cpp, python/sipArcuspart1.cpp, python/sipArcuspart2.cpp, python/sipArcuspart3.cpp, python/sipArcuspart4.cpp, python/sipArcuspart5.cpp, python/sipArcuspart6.cpp, python/sipArcuspart7.cpp

CMakeFiles/python_module_Arcus.dir/build.make:65: recipe for target 'python/sipArcuspart0.cpp' failed
make[2]: *** [python/sipArcuspart0.cpp] Segmentation fault (core dumped)
make[2]: *** Deleting file 'python/sipArcuspart0.cpp'
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/python_module_Arcus.dir/all' failed
make[1]: *** [CMakeFiles/python_module_Arcus.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2
```

ninovanhooff, 2020-01-03:
Try

`sudo pip3 uninstall sip`

```
Uninstalling sip-4.19.8:
  /usr/local/lib/python3.6/dist-packages/sip-4.19.8.dist-info/INSTALLER
  /usr/local/lib/python3.6/dist-packages/sip-4.19.8.dist-info/METADATA
  /usr/local/lib/python3.6/dist-packages/sip-4.19.8.dist-info/RECORD
  /usr/local/lib/python3.6/dist-packages/sip-4.19.8.dist-info/WHEEL
  /usr/local/lib/python3.6/dist-packages/sip.so
Proceed (y/n)? y
  Successfully uninstalled sip-4.19.8

```
I needed to include sudo because without it I got a permission error.

Now I get this result and can start Cura

```
>>> import sip
>>> sip
<module 'sip' from '/usr/lib/python3/dist-packages/sip.so'>
>>> sip.SIP_VERSION_STR
'4.19.18'
```

GregValiant, 2024-11-21:
I'm cleaning house.
Is this still a problem with current Cura versions (5.8.0 and up)?  Can I close this?

pkuiper-ultimaker, 2024-11-22:
Yep, you may close this. Thanks!

GregValiant, 2024-11-22:
Your welcome.

closed 2024-11-22 by GregValiant