Ultimaker/Cura #5932: Cura 4.1.0 Won't start on 2nd attempt

UncleFungus, 2019-06-24:
**Application version**

**Platform**

**Printer**

**Reproduction steps**

**Actual results**

**Expected results**

**Additional information**

UncleFungus, 2019-06-24:
I have Cura 4.1.0 installed on Windows.  I often use Cura in conjunction with Fusion 360.  Usually it opens the first time, but when I go to "Make" subsequently it does not.  I then cannot open it even if Fusion 360 is not running.  It will appear in task manager as cura.exe, but even if I close these instances, it still will not re-open.

nallath, 2019-06-24:
Could you please fill in the bug template and provide the log files?

dieder10, 2019-06-30:
I will fill it because I have the same problem. 
**Application version** - Ultimaker Cura 4.1

Platform - Windows 10

Printer - Anet A8 (that's not the problem)

Reproduction steps - The application only "pops ups" and shows as a window in the desktop the first time it runs after the installation.

Actual results - After the first run, the second run only runs a "Cura.exe" that I can see in the task manager bit any window is shown.

Expected results - The application runs and shows every time I execute it.

Additional information - It happen to me since 4.0.0, so I am working now with the 3.6.0.

Ghostkeeper, 2019-07-01:
Hmm, does it work to run Cura as administrator?

One possible explanation is that the installer runs as administrator and then starts Cura as subprocess as administrator. Then this first run creates the configuration folders as administrator and somehow it's not able to start up again afterwards, perhaps because it's not allowed to write to the configuration folders.

UncleFungus, 2019-07-01:
Thanks dieder10,

I am an administrator & only user of that machine.  I may try un-installing and doing a clean install if that doesn't help. I will roll back to 3.6.

dieder10, 2019-07-01:
Even trying to run it as administrator, only runs the Cura process but not the window. I have tried to reinstalling it, many times. I have realised that if I don't remove the settings when I unistall, the window after installing again it doesn't show up.

Ghostkeeper, 2019-07-01:
The fact that 3.6 works is surprising to me though. Maybe Qt is having difficulty here. Do you guys get a log file?

SimonTSorcerer, 2019-07-02:
Same one for me.

Since a few weeks Cura does not open at all. Before all versions ran nicly. It shows the busy mouse running some circles for 20 seconds but then nothing happens. It could be related to an recent windows update maybe, having new rules set up for programs I suppose. However none of the versions work anymore here. Not even 3.6.0, not 4.0.0, not 4.1.0. If started in parallel, all Cura.exe show up in Task-Manager, but that´s it. Reinstalled several times. Win10 64. All updates installed.

SimonTSorcerer, 2019-07-02:
> Same one for me.
>
> Since a few weeks Cura does not open at all. Before all versions ran nicly. It shows the busy mouse running some circles for 20 seconds but then nothing happens. It could be [...]

Gents, I just found a workaround. It seems to be related to current windows updates as expected. I am sure Ultimaker will release a fix for this.

Simply run compatability manager on Cura.exe. It might set it to Win8 compatability. However it will work with these settings as usual.

Give it a try!

Ghostkeeper, 2019-07-03:
Our IT department regularly makes us install Windows updates but we haven't seen it yet. Maybe we'll see it in a few days then when the next round of updates rolls in.

UncleFungus, 2019-07-03:
Thanks SimonTSorcerer, running Windows compatibility checker fixed the issue for me. Said it wasn't compatible with W10 and automatically applied some sort of fix, so I can stick with 4.1

benJephunneh, 2019-07-05:
Another workaround for me (Win10, Cura v4.1.0) has been to run Cura twice.  Both executions fail, of course, so I end the processes, run Cura a third time, and then it loads successfully.  I'd be glad to provide logs, but I don't know which one you'd want (I see a few in the Roaming directory).

Rolzzz, 2019-07-24:
Compatibility mode worked for me... pretty silly

Nsomnia, 2019-08-22:
Yep happens to me too on a OEM Surface Pro 6, havnt tried on my "usual" desktop rig as thats no where near OEM software if ya catch the wind. 

QT library can be wonky sometimes. Compatibility in win8 or 7 is what seems easiest to get the win10/qt/cura bugs flying together in 1 swarm to the fly swatter, not gonna try building from source and see what I can find out, not these days. Not unless something bigger pops up.

closed 2020-06-19 by no-response[bot]