Ultimaker/Cura #5933: Variable Line Width

Madau3D, 2019-06-24:
I've been using Cura for about a year now and I've been annoyed by a slicing / tool-path situation.  When printing thin walls that don't divide neatly by the selected nozzle diameter or line width, the slicer generates g-code where the nozzle makes a tiny zig-zag movement to fill that space.  This significantly increases print time and is very inefficient use of machine run time.  If it's really bad, I'll either modify my design to get a wall thickness that the machine can print in even lines, or fudge the line width in Cura (or both) until the print time comes down.  This can reduce run time by hours depending on the model.

What I suggest is create a variable line width feature.  When enabled, the user could specify a min/max percentage of nozzle diameter, and the slicer could then decide what line width would best fill an area.  This would make setting perimeter thickness more intuitive as well.  If I am printing with a 0.6mm nozzle, but I want a 2mm wall thickness, currently the slicer has to chose to make the wall thinner (3 perimeters) or thicker (4 perimeters) but the user would not get a 2mm wall in either case.  With variable line thickness the slicer could decide to print a 0.67-68mm line width (I realize the wall thickness isn't an exact multiple of the line thickness as there is some overlap between the lines in a layer) and achieve the specified wall thickness.

I've never tried printing a line width narrower than my nozzle diameter, but I regularly print lines at 125% of nozzle diameter with great results.  With this feature, one could specify, as a hypothetical example, a line width in the range of 95% to 125% of nozzle diameter.

This could be taken a step further, and imagine in the case of printing a narrow tapered features.  The slicer could taper the line width as the the profile narrows to achieve a cleaner part.  There also could be settings for the line width range of inner walls, outer walls, infill and supports- top and bottom skin layers, etc...  Thinner line widths for supports and infill could be of great benefit for reducing filament consumption and possibly making supports easier to remove.

smartavionics, 2019-06-24:
See https://community.ultimaker.com/topic/28556-feature-request-variable-line-width/

Ghostkeeper, 2019-06-26:
Cura has a method to generate lines with variable line width. The problem is pretty hard in general though and we've found that pretty much every solution that works to fix one case will break it for another.

Our current solution is documented in broad strokes here: https://github.com/Ultimaker/CuraEngine/blob/master/docs/generating_paths.md#filling-small-gaps

We have made corrections to make this solution work better in many cases in the 4.2 release, which should take away 90% of the problems in this department. We are also working on a wholly different approach of adjusting the wall line widths, but that won't be realised for some time (a paper is being written about it).

Ghostkeeper, 2020-07-16:
We're working on an alternative solution that generates actual variable line widths for the normal contours: https://github.com/Ultimaker/CuraEngine/pull/1210
This is a huge project though, so it won't get done in 4.7 yet.

Ghostkeeper, 2021-12-14:
This is implemented by https://github.com/Ultimaker/CuraEngine/pull/1210.

closed 2021-12-14 by Ghostkeeper