Ultimaker/Cura #5934: TMC driver mode switching

Jmlannan, 2019-06-24:
TMC stepper drivers allow switching from stealth-chop to spread cycle with 
M569 [E] [I<index>] [T<index>] [X] [Y] [Z]

Currently marlin allows you to set the hybrid threshold M913, but this has some issues. It only switches at high speeds not high acceleration/jerk. Spread cycle is not usually needed while the printer is moving at a high speed, only when its accelerating, allowing for quieter printing and energy saving. 

If Cura could detect when higher jerk/acceleration is needed and switch modes with M569 it would allow for better adaptive printing. All it would require is adding a line before a fast acceleration/jerk is about to occur and turning it off when its no longer needed.

It would also be nice to have a setting to toggle it for the whole print or set M913 for the whole print.

Ghostkeeper, 2019-06-26:
Something for a post-processing script, maybe?

> If Cura could detect when higher jerk/acceleration is needed

When is it needed? If acceleration or jerk control is enabled, then Cura will switch the acceleration and jerk around all the time. Would it be fair to say like "Output M569 whenever setting Jerk to more than 20"? Or does it actually depend on how sharp the corner is?

Jmlannan, 2019-06-27:
The more though I give it the more im thinking this would need to be done on the firmware side. The M569 would need to happen in the middle of a move one the motor is up to speed.

Ghostkeeper, 2019-06-28:
I know that the Ultimaker firmware will automatically drop microsteps as the print speed is increasing, eventually up to full steps.