Ultimaker/Cura #5936: how to pack this  project to Installation package

lishuwei0424, 2019-06-25:
hello ,I want to pack  this project to   Installation package once, how to  do it  ,thank you

rburema, 2019-06-25:
Hi.
I assume you intend to create an installer? What OS do you mean to target?

More generally, build instructions can be found [on these pages](https://github.com/Ultimaker/Cura/wiki/Running-Cura-from-Source) in the wiki. The creation of an installer is a bit out of scope though, since it's more of a general procedure than anything to do with Cura itself.

Ghostkeeper, 2019-06-27:
The result of the CMake script in the Ultimaker/cura-build repository is a package for the operating system you're running (cross-compiling is not supported).

closed 2019-06-27 by Ghostkeeper