Ultimaker/Cura #5937: [4.1.0] Keeps crashing on windows

dnoel8837, 2019-06-25:
**Application version**
4.1.0

**Platform**
Win 10 64bits

**Printer**
Ultimaker 3

**Reproduction steps**
1 - opening Cura
2 - File -> Open   => crash without any error message

or

1 - opening stl file with doucle clic on the file
2 - slicing
3 - trying to save => crash without any error message
[stderr.log](https://github.com/Ultimaker/Cura/files/3324338/stderr.log)

I manage to use Cura only by double clic on an stl file and automaticaly saving the gcode on the portable usb device. 

Any idea how to solve the bugs ? ﻿
[stderr.log](https://github.com/Ultimaker/Cura/files/3324342/stderr.log)

David

rburema, 2019-06-25:
Hi, Thanks for the report.

The crash without error is worrying, it should always at least give an error-report.

In any case, I can't find anything _specific_ in the logs, but the "Cannot find node for machine def [ultimaker3] in QualityChanges lookup table" line at the end makes me think there might be something going on with the definitions. If you don't mind, could you zip the contents of `%appdata%/cura/4.1` an post them here?

dnoel8837, 2019-06-26:
Thank you for your response.

Here is the file you requiered from C:\Users\David\AppData\Roaming\cura
[4.1.zip](https://github.com/Ultimaker/Cura/files/3330270/4.1.zip)

rburema, 2019-06-26:
I coudn't directly reproduce it here with those settings. That it happens on opening files however reminds me of something; do the files you open come from USB, other removable media or dropbox? We've had issues with those in the past, this may be a regression. If so, a quick workaround (for the moment) would be to open local files.

dnoel8837, 2019-06-26:
I can't even get to the point where I select a file to be opened..
I tried to shut down dropbox before opening Cura but nothing has changed.

zajjj, 2019-06-26:
I have the same problem. Windows reinstallation did not help either. I think the problem will be compatible with drivers or CPU.
I have an Aspire ES15 c Celeron N3350 notebook.
Dnoel8837, what hardware do you have?

Ghostkeeper, 2019-06-27:
We've seen this problem before. Something in Qt crashes when opening a file dialogue.

Do you happen to have an Explorer extension, such as Dropbox or Google Drive or something? In the past this has given people some trouble.

dnoel8837, 2019-06-27:
> 
> 
> I have the same problem. Windows reinstallation did not help either. I think the problem will be compatible with drivers or CPU.
> I have an Aspire ES15 c Celeron N3350 notebook.
> Dnoel8837, what hardware do you have?

Hi zajjj,
I have a dell latitude 7450

dnoel8837, 2019-06-27:
> 
> 
> We've seen this problem before. Something in Qt crashes when opening a file dialogue.
> 
> Do you happen to have an Explorer extension, such as Dropbox or Google Drive or something? In the past this has given people some trouble.

Hi ghostkeeper,
I turned of both dropbox and gdrive, nothing has change. Still crashing

Ghostkeeper, 2019-06-28:
Did you also uninstall the Explorer extensions?

IntHelloWorld96, 2019-06-28:
Hey Guys! im having this same problem! 

I have a Dell Inspiron 5000 Series with an intel i7 Processor. Ive attached my %appdata%/cura/4.1 files aswell. 
[4.1_DD.zip](https://github.com/Ultimaker/Cura/files/3340348/4.1_DD.zip)

zajjj, 2019-06-28:
Solved!
the problem is caused by _ssl.pyd.
I deleted it and everything works.
Test and write or help it.

dnoel8837, 2019-06-29:
> 
> 
> Did you also uninstall the Explorer extensions? Turning off the processes is not enough, because Explorer is still running.

Hi Ghostkeeper,
I looked in the program list, I didn't see any extension installed...

dnoel8837, 2019-06-29:
> 
> 
> Solved!
> the problem is caused by _ssl.pyd.
> _ssl.pyd is in C:/Program Files/Ultimaker Cura 4.1/
> I deleted it and everything works.
> Test and write or help it.

I tried the tricks you proposed. Didn't work on my computer..
But thanks

fahimc, 2019-09-09:
I can confirm that, deleting the _ssl file worked. Windows will prompt you to allow cura to access your network. Once you allow Cura then it works fine.

Ghostkeeper, 2019-09-10:
For the next release we've upgraded our Python version to the latest patch version for 3.5. That could solve this issue. Our beta should be released shortly.

closed 2020-06-19 by no-response[bot]