Ultimaker/Cura #5939: allow per model setting on multiple models / intances

enunibw, 2019-06-25:
**Is your feature request related to a problem? Please describe.**
When I'm printing let's say 10 instances of model A and 10 instances of model B in one print Job and want to have different Settings for the models (e.g. infill percentage) I can apply the Standard Settings to model A but the differing Settings for model B have to be applied to each instance seperately. This ist especially cumbersome as the Settings have to be selected out of the full list each time.

**Describe the solution you'd like**
I want to be able to apply per model Settings on a selection of multiple models. Currently this ist greyed out when multiple models are selected

**Describe alternatives you've considered**
None I could think of

**Affected users and/or printers**
Everyone who builds models with different Settings in one print job
**Additional context**

Ellecross, 2019-06-26:
Interesting! :D let me tag @Yian-Lai 

Alternatively, if you will use the same model anyway, you can: 
1. Load model A
2. Modify the per-object-settings of model A
3. Multiply model A

All instances of model A will have the same setting visibility and setting value.

uma-studio, 2021-09-16:
Hi, I have a situation where the feature mentioned would be very useful. Unfortunately I cannot multiply a starting object with the desired per model settings as I need to apply the settings to a group of objects that are not identical. I have tried grouping the objects however the per model settings remain greyed out. I can of course apply them individually but it is not a fast process and definetly seems like it should be done if the objects are at least grouped.

Can anyone advise if there is a timeline towards adding this ability?

Much obliged

cheers

nallath, 2021-09-16:
> Can anyone advise if there is a timeline towards adding this ability?

There isn't a timeline set.

sweiss93, 2022-03-27:
This functionality would be very useful for me as well.

colossatr0n, 2024-11-24:
I would like this as well. For instance, if you have a model with 10 small hinges, then you have to apply overlap settings to a support blocker for each hinge, which is tedious.

Waldehyd, 2025-05-14:
Would love this as well