Ultimaker/Cura #5940: [4.1] Tree Support through model

mike192020, 2019-06-26:
**Application version**
4.1

**Platform**
Windows 10

**Printer**
JGAurora A5

**Reproduction steps**
Add model and slice

**Actual results**
You can see that tree support lines go though model

**Expected results**
No connected lines through model

**Additional information**
Short from writing it all over and submitting files I will point to this post https://community.ultimaker.com/topic/28276-tree-support-failure/?tab=comments#comment-240791

Ellecross, 2019-06-26:
Hey @itisnot-me , unfortunately this ticket is still open. :( sorry .. 

https://github.com/Ultimaker/Cura/issues/5223 

We currently have an internal ticket now in Cura for this. 

Do you mind if we close this keep track of #5223 instead?

closed 2019-06-26 by Ellecross

nallath, 2020-07-01:
Fixed for 4.7