Ultimaker/Cura #5943: Cura 4.1 doesn't save

MechanicalEclipse, 2019-06-27:
So, i use cura a lot, all the other vesion worked well, but today i decided to update to 4.1 .
I setupped my print settings of a project, saved and opened the project the pc connected to the printer... and it was all gone. All the settings and parameters was gone. On the other pc i have the newest version of cura too, but idk why all of my work is gone. Another thing is that the scale function does,t work well and is bugged a lot. Thank you.

Ghostkeeper, 2019-06-27:
Assuming you're on Windows, when you installed Cura 4.1, did you choose to uninstall the older version? And when you uninstalled the older version, did you opt to remove the configuration as well?

If not, could you maybe fill in the issue template? In particular, the log file is important and any error messages you're seeing on the screen.

MechanicalEclipse, 2019-06-27:
Sorry, i didn't sad that i have installed cura on a mac and i uninstalled all the version i had before. I uninstalled and reinstalled cura 4.1 and now i have just 2 problems : 1 on scaling the "support blocker" cube and 2 on saving files(i tried saving 5 projects, only in 2 it didn't saved the settings )

rburema, 2019-07-03:
Just to be sure: You did save (the correct files) with the 'Save...' option in the file menu right? Not the 'Export...' one?

closed 2020-06-19 by no-response[bot]