Ultimaker/Cura #5946: CURA 4.1 crashes when saving

Tuudawgs, 2019-06-28:
**Application version**
CURA 4.1.0

**Platform**
Windows 10 Home, HP laptop

**Printer**
Anycubic i3 Mega,  Monoprice Minidelta 

**Reproduction steps**
Save Gcode to hard drive

**Actual results**
CURA closes, file saved not complete

**Expected results**
Save to hard drive

**Additional information**
Read  issue #4016 and again think it is an HP thing.  The first save following turning the computer on saves successfully but CURA still crashes.  After that files not complete. Tried to find logs in Roaming folder but not there.  I do have the Windows error message.  
Don't know where to find "send file" instruction.

Tuudawgs, 2019-06-28:
Well poop.  I found the "send anonymous" box, unchecked it and it works fine

closed 2019-06-28 by Tuudawgs

Ghostkeeper, 2019-07-01:
Looks like a duplicate of https://github.com/Ultimaker/Cura/issues/5682 then.

Tuudawgs, 2019-07-02:
So it would appear.  I just had no idea where the right box to unceck was.