Ultimaker/Cura #5947: Rotation of 3D View fails using touchpad on Ubuntu 18.04

csetera, 2019-06-29:
**Application Version**
4.0.0

**Platform**
Ubuntu Linux 18.04.2
Dell XPS 15 9570

**Printer**
Ender 3

**Steps to Reproduce**
Can't rotate 3D view using touchpad using the documented CTRL + touchpad combination.  Using the arrow keys does work.  This is true for all models I've attempted.

**Actual Results**
No change in the view

**Expected results**
View should rotate

**Additional Information**
Shift + touchpad properly moves the view, which makes this seem like it may be related to the use of CTRL.  I'd be happy to debug the issue, but I'm not even sure where to start.

Ghostkeeper, 2019-07-01:
> documented CTRL + touchpad combination

Where did you find that documentation? It has never worked that way and is not designed to work this way, so this documentation needs to be edited.

The intended way to rotate the camera using a touch pad is by simulating a right-click drag. On my touch pad that is using a two-finger click and drag.

csetera, 2019-07-01:
I was basing that on https://ultimaker.com/en/resources/52011-hotkeys-controls

I've been unable to find any combination that works for me.  Two-finger click and drag acts as zoom for me.

Ghostkeeper, 2019-07-01:
> Two-finger click and drag acts as zoom for me.

That's what should be the behaviour of the two-finger drag without click.

Ghostkeeper, 2019-07-01:
I'll ask our content team to make a correction to that documentation there.

csetera, 2019-07-01:
When you say two-finger drag with click... you are saying you "click" with two fingers and then hold and move (drag) with those same two fingers?  I can give that a try next time I'm in front of that laptop.

Ghostkeeper, 2019-07-01:
No, I'm using two fingers and press down on the touchpad until it makes a clicking noise. I believe that on my other (Lenovo) laptop it's two fingers and then pressing the button to "click".

csetera, 2019-07-02:
Unfortunately, that still gets me movement of the build volume, but not rotation.  I'm happy to keep digging if you can offer suggestions on what I can do to troubleshoot this.  It sounds like it may be tied to my machine in some way.