Ultimaker/Cura #5948: [4.1] Useless material implementation

Liger0, 2019-06-29:
**Application version**
4.1

**Platform**
Win10 x64

**Printer**
All machines

**Reproduction steps**
Export a material, change the material settings on cura, import the previously exported material.

**Actual results**
The imported material will use the settings of the changed material, and if unlinked it will default to 0.
So the exported material settings will never be used, and thus the material exporting and importing behaviour doesn't make any sense.

**Expected results**
Just use the exported informations, and don't link materials.

**Additional information**
https://youtu.be/Z9XowQOX5rk

fieldOfView, 2019-06-30:
Protip: calling a large part of an application useless because you found a bug does not motivate anyone to fix that bug. Please keep your issue titles descriptive of the issue.

Liger0, 2019-06-30:
The problem is I am unsure if this is intended behaviour or not.