Ultimaker/Cura #5949: [4.1] I can't remove a material, and 2 same materials have different dropdowns

Liger0, 2019-06-29:
**Application version**
4.1

**Platform**
Win10 x64

**Printer**
No matter

**Reproduction steps**
I'll say I can't really reproduce it. It happened while playing with material export/import trying to figure out the bug/useless material implementazion in #5948 .

**Actual results**
I can't remove the silver material. It isn't linked to anything and nothing is linked to it. 
It also isn't used by any extruders and by any machine.
Two dropdowns with the same name are also present

**Expected results**
The material should be removable and there shouldn't be different dropdowns.

**Additional information**

[cura.zip](https://github.com/Ultimaker/Cura/files/3342339/cura.zip)

https://user-images.githubusercontent.com/20385116/60304070-8b8cf580-9938-11e9-87d5-005bada8b695.png

Vandrasc, 2019-07-01:
Hi @Liger0 , thank you for your feedback, unfortunately, if we do not have reliable steps to reproduce then is also hard to fix. Something may have gone wrong when switching between materials and the material was not correctly updated. 
I am also not able to reproduce, I will close the ticket for now but users can still see and use it as a reference when they encounter the same issue. Please re-open if you can reproduce it again.

Many thanks!

closed 2019-07-01 by Vandrasc

Ghostkeeper, 2019-07-01:
The button gets disabled if:
* there is no selected material, or
* the selected material is read-only, or
* the selected material is active in any printer (the current printer or otherwise).

Liger0, 2019-07-01:
Another material is selected, the undeletable material is not active in any printer.
I have no idea about how that could end up being read only.
I shared my full cura folder if that can make you able to set my same instance of cura.

Ghostkeeper, 2019-07-01:
> the undeletable material is not active in any printer.

In your Cura folder it shows that the material you're trying to delete is active in the Creality CR-10S printer.

Liger0, 2019-07-01:
> 
> 
> > the undeletable material is not active in any printer.
> 
> In your Cura folder it shows that the material you're trying to delete is active in the Creality CR-10S printer.

But it isn't so, another material is selected in that machine. :confused:

Liger0, 2019-08-03:
Still same issue in a clean install of 4.2.1. I can't remove some materials even if not used anywhere.