Ultimaker/Cura #5955: [Feature Request] Overhang painting (marker, remover)

Block137, 2019-07-01:
**Is your feature request related to a problem? Please describe.**
A faster and easier way to manipulate support placement than support blocker and custom support.

**Describe the solution you'd like**
Use "brush" to manually identify more areas as overhang by coloring them red in addition to what Cura does automatically from overhang angle.
Also the inverse by removing red overhang identifier.
[image]

**Describe alternatives you've considered**
None

**Affected users and/or printers**
Everyone?

**Additional context**
Example GIF was made in Blender using Weight Paint mode.

fieldOfView, 2019-07-01:
If that were easy to implement, it would have been added a long time ago. Painting on a 3d model is tricky, because it requires a model with very consistently applied uv-coordinates (texture coordinates). Even in the GIF above, you can see it going wrong in the roof overhang on the top (the whole side gets painted all of a sudden).

Ghostkeeper, 2019-07-01:
Actually, we did implement this. It's currently on the `feature_custom_support` branch. It works exactly as Block137 described.

The problem is also implementing the inverse. That would require erasing the previously created support nodes which was a linear algebra challenge. Our aim is still to finish that implementation.

Block137, 2019-07-01:
> Even in the GIF above, you can see it going wrong in the roof overhang on the top (the whole side gets painted all of a sudden).

That's because weight painting paints on vertex instead of face, we'll need to paint on faces.

I know it's difficult to implement, probably need to overhaul several things but I heard they(not devs) like this kind of fancy features.

Ghostkeeper, 2019-07-01:
> That's because weight painting paints on vertex instead of face, we'll need to paint on faces.

Then the current implementation is not exactly as you describe, because the current implementation paints on faces. This is important because large flat overhangs in mechanical models tend to have very few vertices and subdividing is expensive.

Block137, 2019-07-01:
>  the current implementation paints on faces.

That's correct implementation, I meant my example GIF paints on vertices which is not desired.

Nico04, 2022-03-01:
This is a good idea, any progress?

Ghostkeeper, 2022-03-14:
We've not developed this any further recently. But as far as I know the technique I used for that proof of concept still works, so we haven't taken steps back either.

HermanGaviriaMarin, 2022-09-10:
Any progress? It seems that this feature which is wanted by many is not being worked on. Support blockers are awkward to use and do not allow users to place and use supports more intuitively. Please put more priority to the matter, print quality is at an all time high when it comes to CURA, now come other feature improvements such as supports

MountainMaker, 2022-10-14:
I agree with others that This is an important feature and it should be raised on prioritylist

KimmoHop, 2023-01-10:
Would it be possible to have a look at https://ultimaker.atlassian.net/browse/CURA-5482?

Does painting and support generation work on UI/Python side, or does it extend to Cura Engine as well?
How does it generate supports? Does it work, or could it work, kinda like user added custom supports + support blockers, but creating them (approximately) on painted faces?

Or would it be easier to create plugin that somehow combines functionality of custom support pillars and support blockers with repeat fire and possibility to remove them all with single click? (note! I haven't used custom supports and blockers much, add-move-resize cycle feels tedious so forgive me my ignorance ;) )

EklDuderino, 2023-02-22:
The need for paint on supports is still here. I enjoy this feature with Prusa, but much prefer Cura overall. Cura gives me better prints, has a much better interface. The lack of paint on supports at this time is unfortunate. It seems to have been requested for 3 years now. Thank you!

HermanGaviriaMarin, 2023-02-22:
Please Cura devs get pain on supports added. We really have been asking for this feature for a while now. Prusa sliced is making great strides in making a very functional slicer while cura is staying behind

Errandir-of-Gondolin, 2023-06-18:
Halfway through 2023 now. Chiming in that paint-on-supports are really a must have feature at this point in time. I know it's by no means simple, but it's a feature that will push many (me included) to Prusaslicer at some point if Cura doesn't catch up. Support blockers cost way too much time on complex models. I'd like to see this topic receive priority as well, like those above.

jellespijker, 2023-06-18:
we're prepping the stage to enable working on this and much more: https://github.com/Ultimaker/Cura/discussions/15629

cna74, 2024-04-25:
i love cura and i don't want to migrate to prusa or orca. because i like the interface. is there in plan to add this feature after almost 5 years?

EmilEmilchen, 2025-02-20:
Is there any news on this? I would like to use Cura over PrusaSlicer for a few of it's features and the extensibility but I just can not do without this feature!